StephanJoubert/home_assistant_solarman #116: Show total_production when inverter is offline

Geliras, 2022-10-13:
Is there a way for a sensor that will always show the total_production even if the inverter is offline?

honikos, 2022-10-23:
Hi Geliras,
do you mean if your inverter is off by night?
Inverters without battery usually shut down when the panels don't produce electrical power (at night) therefore you won't retrieve data from your inverter because he is offline or the other way you can't check the inverter log, because the build in logger is offline.
Ways to check your total_production even the inverter is offline:
1. connect your inverter to an online monitoring platform (usually provided by your inverter supplier)
2. receive data on a local monitoring platform in your local network (e.g. homeassistant, openhab and smaller solutions). You have to find out if the inverter supplier allows to receive inverter data on a local platform or if the community build a software to do this (decoded the data provided by an interface).

Geliras, 2022-10-23:
Yes thats right, but i found a work-around with input_number

VaillantHassIo, 2022-11-10:
@Geliras mind sharing with us?

distancerunner, 2023-03-14:
I found a workaround. In my configuration.yaml I added a custom sensor template and I added this to the energy dashboard, and removed the solarman sensor. This will change your data and remove your solardata from dashboard:
I think the value **availability**  is the one, which is needed during night times.
```
template: 
    - sensor:
        - name: "Ertrag Pergola Total (kWh) custom"
          unique_id: solarman2mppt_total_production_custom
          unit_of_measurement: kWh
          icon: mdi:solar-power
          device_class: energy
          state_class: total_increasing
          state: "{{ states('sensor.solarman2mppt_total_production')|float(default=0) }}"
          availability: "{{ states('sensor.solarman2mppt_total_production')|is_number }}"
```