StephanJoubert/home_assistant_solarman #117: Help to create parameters file for "ErgoCell" hybrid inverter

fergalish, 2022-10-14:
Hi, this HA integration does not work out-of-the-box for me, and I expect it's because I have this brand of inverter - it seems to be an own-brand for a solar PV installer in Ireland and very difficult to find info online.  So my first question is - does anyone else have this brand of inverter "Ergocell"?

Anyway, to be brief, using the "register_scan.py" python script from pysolarmnanv5 and with a lot of fiddling about and thinking on the results, I have a table of what I believe are input_registers and their corresponding meaning.

e.g. register 3109 corresponds to the daily purchased energy measured in units of 0.1 kWh.  In total I have identified 56 input registers with good confidence and another about 110 unidentified but which return a non-zero value. Many of these 110 seem to never change.

I've also identified 38 holding registers, but their values also don't change and I've not yet figured out what any of those might be.

So now I'd like to create the yaml parameters file for the inverter and I'm starting with the following to extract PV1 Voltage on register 3000.

```
requests:
  - start: 0x0BB0
    end: 0x0BC0
    mb_functioncode: 0x03

parameters:
 - group: solar
   items:
    - name: "PV1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0BB8]
      icon: 'mdi:solar-power'
```

The outcome is that this particular entity has value "Unknown" in HA and all other solarman entities have value "Unavailable". So I guess I'm doing something right, but not enough.

Thanks for any help.

shiftee, 2023-03-08:
I have one. It seems to be a rebrand of the KSTAR Blue S inverter.
See https://github.com/StephanJoubert/home_assistant_solarman/discussions/150

fergalish, 2023-03-08:
Thanks @shiftee . Yes, I believe it's a rebranded Kstar.
I managed to reverse engineer almost all of the inverter registers and created a custom_parameters.yaml file for the solarman integration.
I need to update it with more registers which I have since figured out then I'm happy to share with you.
I'll also check out the yaml file in the thread you linked above - maybe there are more parameters which I couldn't figure out.

frsantos, 2023-03-10:
@fergalish Could you share your `custom_parameters.yml` file? We still have many registers unidentified on the KSTAR, and that could be useful.

fergalish, 2023-03-10:
@frsantos I need to update my custom_parameters.yaml file. It only has about 20 basic registers (power, voltage, current, energy etc). I have many more figured out but I need to add them to the file. The slowest part of it is to remind myself how to write the yaml file, 16-bit Vs 32-bit, signed Vs unsigned, etc.

MrShaneH, 2024-09-25:
Hi @fergalish and @frsantos, I've just started with HA and have the same inverter but you both seem to be much further along with integration. Thought using kstar_hybrid.yaml would be enough but all sensors still return unknown. I'd appreciate any help

frsantos, 2024-09-25:
Hi @MrShaneH 

I don't have a "Ergocell", so I won't be able to help you with the registers. It would be helpful if you can contact support of the brand and ask for the modbus protocol documentation. Some brands have no problems giving it.

Anyway, I moved to https://github.com/davidrapan/ha-solarman integration, which is much more updated, and has similar syntax (it was started taking this integration as a base).

WDLandry, 2024-09-27:
> Hi @MrShaneH
>
> I don't have a "Ergocell", so I won't be able to help you with the registers. It would be helpful if you can contact support of the brand and ask for the modbus protocol documentation. [...]

This was a very timely reply. Thank you so much. Same situation as @MrShaneH and the OP. Using the integration you linked and selecting the K.star has worked perfectly for the Egrocell unit. Thank you again

WDLandry, 2024-09-28:
Hi @MrShaneH, the Ergocell inverter unit I have is coming up as a EC 5000 and battery EC B5. Do you have the same units? The integration is actually picking these numbers up. Everything is returning a value. I was surprised how straight forward this was tbh. I had spent hours playing around previously with Stephen Joubert’s original integration with no luck as well as some other ones and left it a few months back and moved onto other home projects. Started looking again the other day and found my way here. All i did is make sure any other solarman integrations I had installed were deleted before installing the Davidrapan integration frsantos linked.

On the config values I can’t share that at the moment as I’m away from home and don’t have remote access set up but it was pretty straight forward. Dongle IP address, dongle serial and K.Star yaml. Battery voltage and cycle details don’t make a difference as I left default at first before digging out the specs to change later.

For context I’m technically savvy but when it comes to how these processes work I’m very much a beginner. So apologies if my next suggestions are overly basic.
Have you confirmed it is connecting to the dongle? If it’s not it could be as simple as an incorrect IP address or serial number being typed in. Also I have found the solar man dongles finicky to connect to in the past, power cycling the unit has worked for me to resolve this. That said if you’re seeing your data in the solarman app, this shouldn’t be the issue but you never know.

If there’s anything specific you’d like to me check please let me know. I know how frustrating this stuff can be when it you’re trying to resolve it, happy to help.