StephanJoubert/home_assistant_solarman #120: Invalid modbus packets accepted and give corrupted data

steve-cox, 2022-10-16:
Well, not invalid as such - the checksum is validated as correct, but the packet is definitely decoded incorrectly.  But something very strange is going on.... I'll try to explain...

I have one of the 23xxxx loggers (LSW3_15_FFFF_1.0.65) on a Sofar HYD 6000 and I have created a custom profile for it that works correctly.  (If your interested, its just an extended version of the wifikit definition: [sofar_hyd_x000.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/9795376/sofar_hyd_x000.txt))

Every now and then (randomly, it can be 5 minutes, it can be 5 hours) an unusual response packet is received from the logger.  It has a checksum that validates and the logger serial number is correct, but it looks like the logger is missing some of the reply out - comparing with a complete reply shows some similarities.

By enabling the debug logging (and adding some extra to show when a retry was made) I see:

Initial request for 0x200-0x255 (serial blanked) : 
`a5170010450000XXXXXXXX020000000000000000000000000000010302000056c44cd015`

'Invalid' reply:

`a5a900101500fbXXXXXXXX0201414f05007a20000088eb46630103ac000200000000000000000000099b01d20000000000000000137dff8a020ff70e004b0017000100740076000009a600000b1103c4008f0637000000410000000c0000001d00000048039c026a00000018000000110002002000010f140d17febe099b0124000000000000000000000064003100270010000a1ffe00000000000000000000000003730000003a2ee02ee002f10001000000009815`

.... at this point, since the checksum is correct, parsing is attempted.  This for me, without fail, blows up with an exception whilst parsing a string type value.  This is caught, result=0, and  the request (identical to the first) is made a second time and the following response is received:

`a5d500101500f8XXXXXXXX0201064f05003e20000088eb4663000000000000000007c000010000078000030000cd270103ac000200000000000000000000099601ce00000000000000001385ff93020ff737004b0017fffd0070006d000009a200000b1103c4008f0635000000410000000c0000001d00000048039c026800000018000000110002002000010f070d18febc09960124000000000000000000000064003100270010fffb1ffe00000000000000000000000003720000003a2ee02ee002f1000100000000000000000000000007c000010000078000030000cd27c915`

It is 44 bytes longer.  This is parsed without any exceptions being thrown... however the numbers are *way* off.... my 16 panels apparently generate hundreds of kW for a minute.... or battery SoC is >100%.... you get the idea!

The invalid values *always* occur upon a retry.  Thankfully, 99% of the time, a retry is not required and everything works correctly - however it really throws off HA's energy panel, and due to the size of the numbers, the auto scaling on graphs make them impossible to read.

Here is an example of a good reply (sans serial) that was sent without requiring a retry in the minute following the exchange above:

`a5d500101500fcXXXXXXXX0201424f05007a20000088eb4663000000000000000007e400010000073f000200000b620103ac000200000000000000000000099b01d20000000000000000137dff8a020ff70e004b0017000100740076000009a600000b1103c4008f0637000000410000000c0000001d00000048039c026a00000018000000110002002000010f140d17febe099b0124000000000000000000000064003100270010000a1ffe00000000000000000000000003730000003a2ee02ee002f1000100000000000000000000000007e400010000073f000200000b620815`

Now that I have all the packets out of the way, I have some additional info: In the 'parsed incorrectly' failure case above my first string of panels, PV1 (register 0x252) was generating 120,000kW(!).  Now in my case power is scaled by 10, so I guess it can fit in a 16 bit register as 0x2ee0.... the thing is I can find 0x2ee0 in the both of the above packets, so I don't immediately see how it would parse differently upon a retry.

steve-cox, 2022-10-19:
This appears to be working nicely now - I haven't experienced a single corrupted value, and I am getting nice graphs of metric within HA.  I've cleaned up the code ready to submit.

jmccrohan, 2022-10-20:
@steve-cox pysolarmanv5 author here.

Integrating pysolarmanv5 with this library has been on my to-do list for a while, but I haven't gotten around to it.

Rather than duplicate the pysolarmanv5 code within the [solarman.py](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/solarman.py) module, you would be better off using pysolarmanv5's [send_raw_modbus_frame](https://pysolarmanv5.readthedocs.io/en/stable/api_reference.html#pysolarmanv5.PySolarmanV5.send_raw_modbus_frame) method. This will still use home_assistant_solarman's Modbus frame generation algorithm, but will hand off the Solarman V5 element to pysolarmanv5. This will get you the enhanced frame validation referenced above. There are also [further sequence number enhancements](https://github.com/jmccrohan/pysolarmanv5/commit/edeee309191cedb4b16437b66a852276740e6594) sitting unreleased in main. I hope to release these changes with v2.5.0 later this week.

Obviously using the rest of the pysolarmanv5 Modbus methods would be preferable (as it will add support for additional Modbus function codes), but that is a more invasive rewrite.

Regards,
Jon

steve-cox, 2022-10-20:
I did think about doing that initially, but was unsure about how to add required libraries to a HACS integration!  What I have currently implements a subset of the checks and a far simpler (& presumably slower) CRC algorithm.  The net result was actually quite a small set of changes.

I'll look into integrating pysolarmanv5 and make the change the way you suggested instead - it's going to be better solution.

alienatedsec, 2022-10-29:
issue #122 is what you are discussing more in detail - is there any easy fix for this @steve-cox ?

alienatedsec, 2022-10-29:
> Integrating pysolarmanv5 with this library has been on my to-do list for a while, but I haven't gotten around to it.

@jmccrohan maybe your to-do list is going to materialise quicker :) (?)

steve-cox, 2022-10-30:
> issue #122 is what you are discussing more in detail - is there any easy fix for this @steve-cox ?

They do look similar.  It could be that issue 122 is a duplicate of this one.

alienatedsec, 2022-10-30:
Thanks @steve-cox 

I applied the code already - I should see more stable reading from now on.
[image]

alienatedsec, 2022-10-30:
I will also keep a close eye on those superpower anomalies

[image]

alienatedsec, 2022-10-30:
@steve-cox still happening - is there any way to debug it?

[image]

steve-cox, 2022-10-30:
When you applied the fix, did you restart home assistant?  The python wont get reloaded until then.

You can enable debug output [with these steps](https://github.com/StephanJoubert/home_assistant_solarman/issues/53#issuecomment-1284012483)

This will log the raw data to the HA logs.  Maybe there are some additional checks needed, or it could be as simple as needing to do another retry - the integration currently only tries twice.

Then again, the problem may be completely unrelated - you have a Solis inverter, I have a Sofar inverter.

alienatedsec, 2022-10-30:
@steve-cox how often you pull the data in Solarman? Mine is set to 15 seconds. The inverter should not matter much as modbus is modbus.

its been an hour since the last error

[image]

steve-cox, 2022-10-30:
I only pull the data once per minute.  Maybe you are overloading the poor logger!  Like most commercial devices out there, it's probably using the cheapest MCU they could get away with :)

The reason I mentioned our different inverters is because I believe the logger is merely wrapping the unchecked modbus reply from the inverter.  ie - the inverter is the device that it's producing modbus, not the logger.  As such, different inverters could indeed experience similar, but different problems because they each have a different modbus implementation.

If the changes I made didn't fix the problem I saw, one thing I did also consider changing was the addition of a small delay (say 250ms) between retries e.g. [in the current solarman.py](https://github.com/StephanJoubert/home_assistant_solarman/blob/840a6f96432298ef73022314d4d9be386a0aa6a7/custom_components/solarman/solarman.py#L136-L138)

```
from time import sleep
...
...
            if 0 == self.send_request(params, start, end, mb_fc):
                # retry once
                time.sleep(250)
                if 0 == self.send_request(params, start, end, mb_fc):
...
...
```

Maybe that would help you?  I found it useful to add some additional log.debug() around that point so that you know when you are sending the request, a retry, or if the retry has also failed - you may wish to do that too.

alienatedsec, 2022-10-30:
I have enabled debugging and restarted HA after the last error. So far, there has been no error since 11:55, so there is nothing to come back with. I will keep you posted.

[image]

alienatedsec, 2022-10-30:
@steve-cox 
I amended the settings to the same intervals as yours (60 sec), and not much of a behaviour change.
[image]

The logs for that specific time
```
2022-10-30 21:28:48.669 DEBUG (SyncWorker_6) [custom_components.solarman.solarman] a517001045000032c3fcf30200000000000000000000000000000104814c0064180aaa15
2022-10-30 21:28:49.039 DEBUG (SyncWorker_6) [custom_components.solarman.solarman] a5db001015003f32c3fcf30201fe5a09086c1900009291555b0104c800000000000000002af803e800000000000000000000000000010000000000020000000000000000000007010000000000000000000bd2b8095e00b00000000400230210005200010c170964001a002f00641464005107d007d000000000012d0000000001b0000001310000000000000000001e0005000000050000010f0043004c00000121004d004c0000011700480034000001ef000100150000028a009a00930000000001cd000000a50000000000000000000000000000000000000000000000000000000033599d15
2022-10-30 21:28:49.055 DEBUG (SyncWorker_6) [custom_components.solarman.solarman] a517001045000032c3fcf3020000000000000000000000000000010481b00064d83afe15
2022-10-30 21:28:49.339 DEBUG (SyncWorker_6) [custom_components.solarman.solarman] a5db001015004032c3fcf30201ff5a09086c1900009291555b0104c8000108fc138800010001005003e803e801b80217024402580000000f000c000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002095e00b0000000000000000000000004000000000000000000000004000000a90000000000000000000000a9000000a90000000000000000000000a90002138500006d140000c1980000000000000000000000000000000000000000001e001400642ad8dc15
2022-10-30 21:28:55.018 INFO (MainThread) [hole] Response from *hole: 200
2022-10-30 21:29:48.789 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a517001045000032c3fcf3020000000000000000000000000000010480e8006458159015
2022-10-30 21:29:49.131 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a5db001015004932c3fcf302013a5b0908a81900009291555b0104c83105003700a100013630333130353232323632393031333800000000000000000000000000000000000000000016000a001e0015001d002e00000000025b000001d800000083004d00730000025b0000000000000000000000000000000000000000001a0001001600010000000000000000000000000000000000000000000000000000000000000000000200000f430000096800000000001a000000000000012cfffffff6000001220000000000000000000000000000003500fa13890003f82f0000000800004caa3315
2022-10-30 21:29:49.150 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a517001045000032c3fcf30200000000000000000000000000000104814c0064180aaa15
2022-10-30 21:29:49.234 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a5db001015004932c3fcf302013a5b0908a81900009291555b0104c83105003700a100013630333130353232323632393031333800000000000000000000000000000000000000000016000a001e0015001d002e00000000025b000001d800000083004d00730000025b0000000000000000000000000000000000000000001a0001001600010000000000000000000000000000000000000000000000000000000000000000000200000f430000096800000000001a000000000000012cfffffff6000001220000000000000000000000000000003500fa13890003f82f0000000800004caa3315
2022-10-30 21:29:49.249 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a517001045000032c3fcf3020000000000000000000000000000010481b00064d83afe15
2022-10-30 21:29:49.632 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a5db001015004a32c3fcf302013b5b0908a81900009291555b0104c800000000000000002af803e800000000000000000000000000010000000000020000000000000000000007010000000000000000000bd2b8096200b1000000050023020f005200010c17096a001a002f00641464005107d007d000000000012b0000000001b0000001300000000000000000001e0005000000050000010f0043004c00000121004d004c0000011700480034000001ef000100150000028a009a00930000000001cd000000a50000000000000000000000000000000000000000000000000000000032e2b215
2022-10-30 21:30:10.101 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solisv2_daily_energy_exported, value dropped from 0.1 to 2022-10-30T21:29:49.704013+00:00, triggered by state with last_updated set to 0.0
2022-10-30 21:30:10.101 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solisv2_daily_house_backup_load_consumption, value dropped from 15.4 to 2022-10-30T21:29:49.712240+00:00, triggered by state with last_updated set to 0.0
2022-10-30 21:30:10.101 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solisv2_total_house_backup_load_consumption, value dropped from 650.0 to 2022-10-30T21:29:49.716840+00:00, triggered by state with last_updated set to 0.0
2022-10-30 21:30:10.101 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solisv2_total_battery_charge, value dropped from 271.0 to 2022-10-30T21:29:49.757468+00:00, triggered by state with last_updated set to 0.0
2022-10-30 21:30:10.101 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solisv2_total_battery_discharge, value dropped from 289.0 to 2022-10-30T21:29:49.759951+00:00, triggered by state with last_updated set to 0.0
2022-10-30 21:30:48.393 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a517001045000032c3fcf3020000000000000000000000000000010480e8006458159015
2022-10-30 21:30:48.731 DEBUG (SyncWorker_0) [custom_components.solarman.solarman] a5db001015006032c3fcf30201765b0908e31900009191555b0104c83105003700a100013630333130353232323632393031333800000000000000000000000000000000000000000016000a001e0015001e002e00000000025b000001d800000083004d00730000025b0000000000000000000000000000000000000000001b0001001700010000000000000000000000000000000000000000000000000000000000000000000200000f470000096f00000000001b000000000000012c000000000000012c0000000000000000000000000000003500fa138c0003f82f0000000800002de30315
```

steve-cox, 2022-11-01:
FYI: I'm not a whizz on this stuff, I've just read through [jmccrohan](https://github.com/jmccrohan)'s awesome pysolarmanv5 library and seen how it does things...

Taking the (presumably bad) response received at 21:29:48.789, I can see that the control code is 0x1015 as expected, and the frame type is 0x02 also as expected:
`a5db00`**`1015`**`004932c3fcf3`**`02`**`013a5b0908a81900009291555b........`

Also, the v5 checksum is correct (it's correct on all the responses).

Next up is the embedded modbus frame.  I've taken the 3 responses received at 21:29 and run them through [a decoder](https://rapidscada.net/modbus/) (use "Modbus RTU" and "Response"), and they all seem to decode just fine - valid CRCs etc.  The process is pretty straightforward:

I trim off the first 25 bytes and the last 4 bytes in order to get to the modbus frame embedded within the v5 frame:
`a5db001015004932c3fcf302013a5b0908a81900009291555b `**`0104c83105003700a100013630333130353232323632393031333800000000000000000000000000000000000000000016000a001e0015001d002e00000000025b000001d800000083004d00730000025b0000000000000000000000000000000000000000001a0001001600010000000000000000000000000000000000000000000000000000000000000000000200000f430000096800000000001a000000000000012cfffffff6000001220000000000000000000000000000003500fa13890003f82f0000000800004caa`**` 3315`

(BTW: easy to find the first 25 bytes/50 chars - **your** logger will always have 9291555b at that position - its the serial!)

Running that middle portion through the online decoder mentioned previously gives me:
Part of Data Package | Description | Value
-- | -- | --
01 | Slave address | 0x01 (1)
04 | Function code | 0x04 (4) - Read Input Registers
C8 | Byte count | 0xC8 (200)
31 05 00 37 00 A1 00 01 36 30 33 31  30 35 32 32 32 36 32 39 30 31 33 38 00 00 00 00 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 00 16 00 0A 00 1E 00 15 00 1D 00 2E 00 00 00 00  02 5B 00 00 01 D8 00 00 00 83 00 4D 00 73 00 00 02 5B 00 00 00 00 00 00  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1A 00 01 00 16 00 01 00 00  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 02 00 00 0F 43 00 00 09 68 00 00 00 00 00 1A 00 00  00 00 00 00 01 2C FF FF FF F6 00 00 01 22 00 00 00 00 00 00 00 00 00 00  00 00 00 00 00 35 00 FA 13 89 00 03 F8 2F 00 00 00 08 00 00 | Register value | 0x3105 (12549), 0x0037 (55), 0x00A1  (161), 0x0001 (1), 0x3630 (13872), 0x3331 (13105), 0x3035 (12341),  0x3232 (12850), 0x3236 (12854), 0x3239 (12857), 0x3031 (12337), 0x3338  (13112), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0),  0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0016 (22),  0x000A (10), 0x001E (30), 0x0015 (21), 0x001D (29), 0x002E (46), 0x0000  (0), 0x0000 (0), 0x025B (603), 0x0000 (0), 0x01D8 (472), 0x0000 (0),  0x0083 (131), 0x004D (77), 0x0073 (115), 0x0000 (0), 0x025B (603),  0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0),  0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x001A (26), 0x0001 (1),  0x0016 (22), 0x0001 (1), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000  (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000  (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000  (0), 0x0002 (2), 0x0000 (0), 0x0F43 (3907), 0x0000 (0), 0x0968 (2408),  0x0000 (0), 0x0000 (0), 0x001A (26), 0x0000 (0), 0x0000 (0), 0x0000 (0),  0x012C (300), 0xFFFF (65535), 0xFFF6 (65526), 0x0000 (0), 0x0122 (290),  0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0), 0x0000 (0),  0x0000 (0), 0x0035 (53), 0x00FA (250), 0x1389 (5001), 0x0003 (3),  0xF82F (63535), 0x0000 (0), 0x0008 (8), 0x0000 (0)
4C AA | CRC | 0x4CAA (19626)

All three responses received during 21:49 decode correctly and have the correct checksum.

Unfortunately, this bug related to invalid modbus frames (bad CRC) being accepted, so I'm not sure this is the exact same problem as you are encountering... the responses I've tested all seem to decode correctly.

If you run your responses through the online decoder, do the received register values make any sense? (the response simply requests them sequentially, from start to end register as defined in your inverter definition file).

It might be worth opening up a new bug for the problem you have.  I'm pretty certain the bug I raised here is a different problem, so it's best not to confuse the two - although I'm happy to be corrected.

bbjunkie, 2022-11-01:
@alienatedsec Seeing the same issues here since installing the integration yesterday. Prior to this I was using solismon3 script by NosIreland  (incidentally I see you have forked it so most likely have plenty of experience there) It would randomly throw errors but not get through to HA. Gradually it would get worse, until at one point I needed to restart the inverter. May be worth trying a restart to at least narrow down where the errors are entering the data chain.

alienatedsec, 2022-11-02:
@steve-cox thanks for the 'whizz' analysis :) I know how to approach it now so I will take it from there.

@bbjunkie yes, I am using solismon3 on a separate Linux VM, in combination with Node-RED and MariaDB, to produce the below. However, many things are still 'whizz' level to me, and I'll need to educate myself to get my head around the issue we are experiencing with the solarman integration.

[image]

closed 2022-11-09 by StephanJoubert

sddunne, 2022-12-10:
> #122

Thanks Steve! Spent all night searching for a solution to getting the data in following an install this week and this mostly works brilliantly on my HYD-6000 -- am seeing the same issue as you re the random spikes of data, the Solar Total Generation spikes to 327,680,000 kWh every few mins! What are the code changes you made to fix this, couldn't see them in the comments? Keen to give them a bash!