StephanJoubert/home_assistant_solarman #121: for doc: SN 40..... and old WIFI Stick works also. Solis Inverter RHI-3P K-HVES-5G

honikos, 2022-10-18:
Hi Stephan,
could you update the documentation?
My WIFI-logger has FW MW3_15_0501_1.18 and SN starts with 40........
This worked out of the box.
Thanks!

Janthenman67, 2022-10-25:
Hi Honikos, could you give me a few pointers? How did you get it to work? I also have a Solis Inverter with stick SN starting with 40, but unable to get it to work so far.

Janthenman67, 2022-10-25:
> Hi Stephan, could you update the documentation? My WIFI-logger has FW MW3_15_0501_1.18 and SN starts with 40........ This worked out of the box. Thanks!

Hi Honikos, could you give me a few pointers? How did you get it to work? I also have a Solis Inverter with stick SN starting with 40, but unable to get it to work so far.

honikos, 2022-10-27:
> Hi Honikos, could you give me a few pointers? How did you get it to work? I also have a Solis Inverter with stick SN starting with 40, but unable to get it to work so far.
Maybe it's a combination of inverter model an logging-stick. This was my way:
1. try to find out if you will get any data form your logger. This is documented on https://github.com/jmccrohan/pysolarmanv5
2.  Install and run the first script from Example https://pysolarmanv5.readthedocs.io/en/stable/ on your machine. If you will receive an answer, your are able to receive data (replace the example data in the script, e.g., your logger stick serial number)
3. details see https://github.com/jmccrohan/pysolarmanv5

The second example script didn't work on my machine / gave no output. Maybe the first script opened a channel or function on the logger, the way, that it is allowed to receive answers in your/my local network?

mhspelt, 2022-10-28:
Hi @honikos I tried the steps above, verified that the stick is listening on port 8899. The _scan_ script does indeed find it. The first example script returns an IllegalDataAddressError on every read. And the sensors in HA still all return _unavailable_. 

I actually think that the Omnik integration I am using is working fine. The only problem is that the web UI only returns the total yield in whole kWh so I get weird jumps, whereas the yield for today does have decimal figures.

mhspelt, 2022-10-28:
Here is my config. Which yaml config did you select?
[image]

honikos, 2022-10-28:
> Here is my config. Which yaml config did you select?
due i have a solis hybrid inverter, I use solis_hybrid.yaml

solarjones, 2022-10-28:
A call for some assistance! I've cloned both pysolarmanv5 & uModbus from Git into HA Code Editor, but can't run the examples. Getting errors where it can't resolve the imports: "ModuleNotFoundError: No module named 'pysolarmanv5'".  I'm sure I've missed something, but its beating me currently. Thanks in advance.

honikos, 2022-10-29:
>  I've cloned both pysolarmanv5 & uModbus from Git into HA Code Editor, but can't run the examples.

There is no need to run pysolarmanv5 on your HA machine. Or better: It depends on how your run your HA (HA operating system? container? manual installation)

### Installation on a local machine

**Installation  (from https://github.com/jmccrohan/pysolarmanv5 )**
https://github.com/jmccrohan/pysolarmanv5#dependencies

To install the latest stable version of pysolarmanv5 from PyPi, run:
`pip install pysolarmanv5`

To install the latest development version from git, run:
`pip install git+https://github.com/jmccrohan/pysolarmanv5.git`

Then run a script (I uses first script, replace the logger information with this from your local logger stick) form example page in documentation:  https://pysolarmanv5.readthedocs.io/en/stable/   --> the "Basic Client Example"

If you receive an useful answer, you are able to receive data from your local logger stick.

Now I installed the HA integration via hacs:
https://github.com/StephanJoubert/home_assistant_solarman

Maybe useful:  https://github.com/StephanJoubert/home_assistant_solarman#auto-discovery

solarjones, 2022-10-30:
I've installed HA on a VM on a Synology NAS, so its the full HA OS.

HACs install of the Solarman integration doesn't produce any telemetry (ports opened etc) but reading the various threads there are many different versions of the data logger stick. I have the LAN version, and think the MODBUS encapsulation in the TCP stream is different, hence why the Solarman integration isn't working.

I've replaced the IP address + S/N in the example scripts of Pysolarmanv5, but the issue is the Python environment under HA. This is the Terminal output when running the py example HA:

➜  pysolarmanv5 git:(main) ✗ /usr/bin/python /root/pysolarmanv5/examples/register_scan.py
Traceback (most recent call last):
  File "/root/pysolarmanv5/examples/register_scan.py", line 2, in <module>
    import PySolarmanV5
ModuleNotFoundError: No module named 'PySolarmanV5'
➜  pysolarmanv5 git:(main) ✗ 

So I somehow need to configure something (PATH?) in Ham but unsure what! 

Thanks again for your continued help.

honikos, 2022-10-31:
> but the issue is the Python environment under HA.

so don't use HA. Use another standard machine with installed appropriate python.
You can check with  "python3" --> "modules" if module pysolarmanv5 is available.

alienatedsec, 2022-11-02:
@solarjones you can try this repo https://github.com/jmccrohan/ha_pyscript_pysolarmanv5

It didn't work for me so far. See https://github.com/jmccrohan/ha_pyscript_pysolarmanv5/issues/1

solarjones, 2022-11-11:
@alienatedsec 
Found this [https://github.com/wills106/homeassistant-solax-modbus](https://github.com/wills106/homeassistant-solax-modbus) intergration.

Works a treat with my Solis LAN & RHI-6K-48ES-5G inverter for displaying 90+ entities. Not tried the charging functionality yet.

alienatedsec, 2022-11-11:
> @alienatedsec Found this https://github.com/wills106/homeassistant-solax-modbus intergration.
> 
> Works a treat with my Solis LAN & RHI-6K-48ES-5G inverter for displaying 90+ entities. Not tried the charging functionality yet.

I have solis wireless stick, and it seems not to work correctly as no entities are found after the config. What is your config like? @solarjones

solarjones, 2022-11-11:
My LAN stick output is on port 30003 (default). Not using Server B. The Solax integration config is using TCP, modbus address =1, port=30003 and selected solis_hybrid. I simply installed the Integration and it worked (after updating to todays version). No manual config at all.

alienatedsec, 2022-11-11:
> My LAN stick output is on port 30003 (default). Not using Server B. The Solax integration config is using TCP, modbus address =1, port=30003 and selected solis_hybrid. I simply installed the Integration and it worked (after updating to todays version). No manual config at all. I did have to rip out all other modbus related integrations as i think they were clashing.

@solarjones what about stick serial number details? are those needed on yours?

solarjones, 2022-11-11:
No stick serial number needed with Solax integration. It looks at the inverter serial number automatically to determine which elements it uses. Very easy.

EmbassyGroup, 2022-11-13:
I want to use solarman stick logger with deye inverter to display battery status, grid input...etc on my website but i haven't done it before how can i go about it????

honikos, 2024-07-16:
is changed in docs

closed 2024-07-16 by honikos