StephanJoubert/home_assistant_solarman #128: HYD 6000 EP

lugio, 2022-11-04:
Hi Stephan,
nothing for HYD 6000 EP :(
Thanks

lula73, 2022-12-09:
hyd 6000  zcs brand (zucchetti)  use : 
[solarman-ZCS_HYD-6000.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/10195413/solarman-ZCS_HYD-6000.txt)

fcrohas, 2023-01-05:
Hi,

  @steve-cox i've also a sofar HYD6000 EP, so i tried your file and the patch, but still nothing, status state is still deconnected.
  Maybe it is related to my data logger version the SN start with 236xxxxxxx  
  I checked on the evening with simple command line nc <ip address> 8899 and i receive data frame .. I ve to check hexadecimal data maybe

fcrohas, 2023-03-08:
OK fixed inlatest release.
I have done some test with pysolarmanv5 and found the correct register.
In version 1.4.3 i see that sofar_g3hyd.yaml have the right register reads for the HYD 6000 EP it works well register start at 0x400

It now works with hass integration ... Still some bad value reading sometimes but it works !

steve-cox, 2023-03-09:
@fcrohas Someone else added a HYD 6000 EP config file (sofar_hyd3k-6k-es.yaml) in the latest release - did you try that one?

It's the one I'm using now, as I'd rather keep a standard configuration.  It has 'SOB' (state-of-battery 0-100%) incorrectly labelled as 'Battery Power', but otherwise seems okay.

lepalom, 2023-07-23:
@steve-cox I have a HYD 5000 EP with updated firmware and that file is no working. Using sofar_g3hyd.yaml, I obtain some data, but my inverter is single phase. Any idea?