StephanJoubert/home_assistant_solarman #136: Solis Datalogger SN 192XXXXXXX

everitj, 2022-11-21:
Hi, I am unable to get this work with datalogger SN as above. Using the pysolamanv5 scan script on a pi I am able to find the datalogger so assume it must be compatible. I have tried port 8899 and 30003 but can't pick any data up from the inverter. Datalogger is hard wired type. Any pointers. Thanks

WilLlLli, 2022-11-24:
I had the same issue and as far is I found out, solarman integration is only working with wifi sticks. I have both and have it working  with the wifi stick. It would be nice to see Solarman integration working also with the LAN stick anyway...

everitj, 2022-11-25:
Thanks, that's a bummer, I wonder why.. Has anyone had sucess with the ethernet logger?

alienatedsec, 2022-12-22:
Try this @everitj https://github.com/wills106/homeassistant-solax-modbus