StephanJoubert/home_assistant_solarman #145: Modbus Slave ID?

Klagio, 2022-12-02:
Hi, I do have 2 SUN-12K-SG04LP3-EU one Master and one Slvae.

According to the manual   1 is the Master and 2 is the Slave. Each has a logger, so what do you think I need to add another solarman component instance?

192.168.1.253 where I put 1 (the Logger attached to the Master)
192.168.1.254 where I put 2 (the Logger attached to the Slave)

[image]

[image]

photogeen, 2023-03-17:
Hi, did you find a solution to add two inverters in parallel?

francis91joseph, 2023-04-18:
Hi, I have the same issue. Can someone help?

Privatecoder, 2023-11-08:
just leave it at `1` for all your inverters and add them regularly, however you need to _fix_ the inverter definition if you hit the [`-100` _degree battery-temp issue_](https://github.com/StephanJoubert/home_assistant_solarman/issues/181#issuecomment-1802120189)