StephanJoubert/home_assistant_solarman #160: Connection problem using S/N starts with: 1PH HYD6000 ZSS HP

Simonik71, 2022-12-30:
Hi all
I have an inverter with S/N starting with: 239xxxx.
The script doesn't work, i get: "Solarman status_connection" -> "disconnected".
Could you plese send me any suggestion ?
Regards

MartinMcFlyITA, 2022-12-31:
I have a zucchetti ZCS HYD6000 ZSS HP inverter
to export the data use a solarman datalogger
with serial number 270XXXXXXX and with firmware  version LSW3_15_270A_1.53
on HA I see 77 entities, but all without data and it is not connected.

How can I do to obtain data from my solarman datalogger to my HA installation? 

kind regards

socapex, 2022-12-31:
Hi,
have the same ZCS HYD6000 ZSS and same datalogger with serial number 270XXXXXXX and firmware version LSW3_15_270A_1.53 on HA I see 77 entities, but all without data and it is not connected. Inside the status_connection sensor every time that restart HA i see the connection only for 30 secs, but not value for other sensors. Thanks in advance

paobrunelli, 2023-01-01:
Hi,
same problem with inverter zucchetti ZCS HYD5000 HP. In the log file I find this error:

Traceback (most recent call last):
  File "/config/custom_components/solarman/__init__.py", line 35, in update_listener
    hass.data[DOMAIN][entry.entry_id].config(entry)
KeyError: 'solarman'
2023-01-01 22:06:24.506 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on /homeassistant/restart request Server disconnected

Oiligriv, 2023-01-02:
Hello
I also own a ZCS Inverter, my model is ZCS HYD6000 HP.
I think the HP series stands out from the current homeassistant yaml setup.
Another note, if I connect to the wifi dongle web page I don't see the inverter information, I don't know if it will help.

We hope that it will be possible to integrate our ZCS HP sooner rather than later.

thanks again for what you do

paobrunelli, 2023-01-02:
In the while I have posted a request to ZCS support, asking for info required to setup the integration with the HP serie inverter. Let's see...

Simonik71, 2023-01-04:
> In the while I have posted a request to ZCS support, asking for info required to setup the integration with the HP serie inverter. Let's see...

Could you please share the request, so we can do the same (as pushing...)
Regards

TommasoPino, 2023-01-04:
In the meanwhile, I tried to get a registry map for the **1PH HYD6000 ZSS HP** inverter.
The attached CSV file is the function code 03 registries map, with the address, the value, and the hex value. I do not know how to use it to create a YAML configuration file. Is there anyone that could help with it?
[registriesMap.csv](https://github.com/StephanJoubert/home_assistant_solarman/files/10346170/registriesMap.csv)

paobrunelli, 2023-01-05:
these are the email exchanged with ZCS Support, after they sent me credentials to access their server
- my mail:
Buon giorno e buon anno.
Grazie per la disponibilità.
Il mio obiettivo è quello di integrare i dati estratti dall’inverter in scenari/automazioni Home Assistant per la gestione ottimale dei carichi in funzione dello stato dell’impianto PV. A questo scopo il considerevole ritardo con cui i dati sono resi disponibili dal vostro server costituisce un problema. C’è modo di ridurre drasticamente tale ritardo? Osservo che è uno standard per i portali dei fornitori di soluzioni IoT la disponibilità dei dati in tempo reale. Certo, si tratta di componenti molo più semplici di un inverter, ma i volumi di misure trattate mi aspetto siano paragonabili; io ho un solo inverter e una ventina di dispositivi IoT, per i quali, ripeto ho dai portali dei fornitori dati in tempo reale. 
Nel cercare un’alternativa, ho visto in GitHub un package (https://github.com/StephanJoubert/home_assistant_solarman ) che consente l’accesso diretto, in tempo quasi reale, ai dati dell’ inverter attraverso l’interfaccia wifi LSW 3, senza quindi dove accedere al vostro server.  Nel package sono disponibili i file di parametrizzazione per l’accesso alle variabili di diversi tipi di inverter che utilizzano l’interfaccia wifi LSW 3, fra cui il vostro HYD 6000-ZSS. Ho provato ad utilizzare il file nell’integrazione home assistant verso il mio inverter HYD 5000-ZSS HP, scontando qualche problema sulle variabili, ma l’integrazione HA non riesce nemmeno a collegarsi all’inverter. 
Vi arei grato se poteste fornirmi indicazioni/documentazione utili a far funzionare detta integrazione.
Grazie ancora per il supporto.
Un cordiale saluto
Paolo Brunelli
- ZCS Answer:
Buongiorno,
alcuni clienti hanno richiesto il nostro protocollo modbus al fine di comunicare in tempo reale con l'inverter.
Questa è l'unica soluzione disponibile ed autorizzata da ZCS.
Se lo desidera può richiederlo gratuitamente, col solo impegno di firmare un NDA.
- my further comment:
Grazie per la pronta risposta, se mi inviate l’NDA da firmare provvedo a rinviarvelo firmato.
  
Nessuna prospettiva per una più alta frequenza di aggiornamento dei dati sul server? …se ci fossero problemi di consistenza su un dataset esteso (ipotizzo…) è possibile individuare un sottoinsieme di dati validi “istantaneamente”, associati ad un timestamp?
Grazie ancora
Paolo brunelli
- final ZCS answer:
Buonasera,
purtroppo non siamo strutturati per fornire servizi così specifici.
Le allego l'NDA concordato.

Oiligriv, 2023-01-05:
Ciao @paobrunelli 
Avevo trovato qualcosa online su modbus, ma è complicato da costruire e configurare.
Speriamo che qualcuno riesca a trovare una soluzione sul yaml di configurazione. 
Rimango deluso da ZCS che non sia strutturata per lo sviluppo di un portale con dati in tempo reale. 

Virgilio

paobrunelli, 2023-01-05:
i can't imagine any real technical reason for zcs  not providing real time data out from their sever, nor supporting any real time interface toward the inverter. Such interfaces would be the basis for any serious energy management solution.

TommasoPino, 2023-01-05:
I am completely on you side. I also totally do not understand why a simple
API has not been hosted on the usb Stick attached to the inverter. At this
time, it is the extremely simple to implement and useful to use for the IoT
world. Bah

If it is possible for you, could you share the information privately? Could
help me to complete the reverse engineering ;)

Thanks

TommasoPino, 2023-01-06:
> Hi all, I've been able to integrate the inverter into HA using the latest
> version of the Solarman extension.
> You need to re-download the extension using the latest version available.
> You need to enable "beta versions" and then select "main".
> It worked for me.
> Hope this helps

Which configuration file have you used?

fulippo, 2023-01-06:
> Which configuration file have you used?

sofar_hyd3k-6k.yaml

Simonik71, 2023-01-06:
Unfortunately, doesn't work for me. Are you already connected to the grid to sell power ?

paobrunelli, 2023-01-06:
it does not work form me neither!
but with a small step ahead: now the integration sheos in HA the same files the are in github, previously I was missing somee inverter confiuration

TommasoPino, 2023-01-07:
> Hi all, I've been able to integrate the inverter into HA using the latest version of the Solarman extension. You need to re-download the extension using the latest version available. You need to enable "beta versions" and then select "main". It worked for me. Hope this helps

Are you sure to have the **HYD 6000-ZSS HP** and not the **HYD 6000-ZSS**?

By the way, it does not work for me also. The problem is that the entities are at different registry addresses

MartinMcFlyITA, 2023-01-07:
using "main" branch  (inverter zucchetti ZCS HYD6000 ZSS HP  and datalogger serial 270XXXXXXX )

I see
2023-01-07 22:07:13.349 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry 192.168.88.11 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 533, in async_unload
    result = await component.async_unload_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
    hass.data[DOMAIN].pop(entry.entry_id)
KeyError: 'solarman'

sofar_hyd3k-6k.yaml setting loaded

M

TommasoPino, 2023-01-13:
I have tested it with sofar_g3hyd.yaml and it perfectly works ❤️

paobrunelli, 2023-01-16:
Yes for me as well! Sofar_g3hyd works perfectly with zcs PH HYD5000HP.  now I have 60+ measures available, but for some of them it's difficult (for me) to understand what do they mean. Does any one have documentation on zcs inverter measures? Thanks a lot for any help

TommasoPino, 2023-01-16:
Hi paobrunelli, which value do you not understand?

socapex, 2023-01-16:
Hi All ! sofar_g3hyd.yaml work great, just rename the only fase in use and comment the other two fases.
in the future and with a bit of time will compare the sofar_g3hyd.yaml and sofar_hyd3k-6k.yaml and relative registers to try to find some missing value., 77 instead of 61. Many Thanks to All

paobrunelli, 2023-01-17:
hi Tommaso thank you for your help! I don't understand why I have three times actve power output, current output, voltage phase and current PCC, once ..._S, the _T and _R

thank you again

paolo

socapex, 2023-01-17:
@paobrunelli, Ciao Paolo, because the file is write for the 3 phase inverter, R.S.T, in mono phase you have only one, the R, so in HA entity menu you can delete or disable the ones you don't need.

Repo95, 2023-02-19:
Some of the values are updating, but for others, like daily battery charge and daily battery charge, the the value is not updating. Do you have any idea why this happen?
I have ZCS HYD6000 ZSS HP,  and logger: 270xxxxxxx LSW3_15_270A_1.53

kavejo, 2023-02-27:
I do have the same problem here.

My Inverter is an AZZURRO 3PH HYD6000 ZSS and I have tried to set that up with the sofar_hyd3k-6k-es.yaml configuration file.
I have a Connext linked to the inverter, but given there was no integration in Home Assistant available, I also purchased a Wi-Fi USB stick with firmware LSW3_15_270A_1.53.

If I use the SolarMan Smart app I can see the data (from the logger) - very much in line with what I can see via the AzzurroSystem app (from Connext), however I don't seem to be able to get any data in Home Assistant via this integration.

In the log I can see the following warning:
```
Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:173
Integration: Solarman (documentation, issues)
First occurred: 15:23:29 (10 occurrences)
Last logged: 15:36:20

Querying registers [512 - 597] failed, aborting.
```

I can also see the following error:
```
Logger: homeassistant.config_entries
Source: custom_components/solarman/__init__.py:28
Integration: Solarman (documentation, issues)
First occurred: 15:30:52 (1 occurrences)
Last logged: 15:30:52

Error unloading entry 10.3.0.1 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 533, in async_unload
    result = await component.async_unload_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
    hass.data[DOMAIN].pop(entry.entry_id)
KeyError: 'solarman'

```

Would anybody be able to help me resolving this issue?

Thanks!

TommasoPino, 2023-02-27:
I think you have to use Sofar_g3hyd.yaml file for that. It works like a
sharm for the 1PH but it has the registry for other two phases

Let us know

kavejo, 2023-02-27:
Thank you @TommasoPino. Let me give it a try and report back, however that would be strange.

If I look at the readme.md on https://github.com/StephanJoubert/home_assistant_solarman I can see the following:
[image]

Mine is exactly a HYD6000 ZSS, so according to the documentation I would be using the correct file already.
Also, if I search an image of the inverter for the file you are suggesting (Sofar KTLX-G3), that looks very different form mine.

TommasoPino, 2023-02-27:
> @TommasoPino <https://github.com/TommasoPino> - sadly changing the
> configuration has not resolved the problem.
>
> I have reached out to ZCS support and they said that unfortunately modern
> inverters do not allow data to be queried locally, which is strange as I
> can read on the Home Assistant forum that other users with the same antenna
> are getting the data just fine.

Do not listen the ZCS support… they are extremely capable of passing sheet
from sofar to us… i would like to see a photovoltaic part without querying
data locally… bah

After that in the afternoon I will post here a python script that I used to
map the registry directly. Maybe it could help you debugging and solve this
situation.

kavejo, 2023-02-27:
When I chased them up, they said that they only support querying the data directly via ModBus RTU / RS485.
They shared these protocol specifications, not sure if they could be useful to query data from inverters like mine.

[ModBus-RTU Communication Protocol 3PH Hybrid and Inverter.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/10841199/ModBus-RTU.Communication.Protocol.3PH.Hybrid.and.Inverter.pdf)

Pigi-102, 2023-03-27:
Hello, just joined the discussion as I have just started my Azzurro (1PH HYD 6000 ZSS HP ) with wifi sernum 239XXXXXXX.
I can confirm that if I use sofar_hyd3k-6k-es.yaml I get 77 entities, but never get connected with the same errors mentioned from other collegues.
If instead I use sofar_g3hyd.yaml I get a good connection and 180 entites.Quite a bunch of this are unavailable but other are available and with values, not sure if good or bad but at least values.
Strange thing I cannot see the 2nd battery ( says unknown for all values ), but this could also be a problem in installation ( not done by me ).
Battery 1 temperature jumps from  2,279 °C, which looks quite high to me, and 0°
Battery 1 voltage jump from 50ish and 227ish that seems strange too.

Ambient temperature too is strange saying 43° but inverter and battery are in a box without warming so these days could be around 15, no more than this.

Other stuff I see are all the vaules ending in "T" "R" and "S" are unknown or 0, but the "Total"s are fine.

Voltage_Phase_R is fine.

Pigi-102, 2023-03-28:
@kavejo No way the "sofar_hyd3k-6k-es.yaml" works in any way.
It expose these 77 entities but it remains disconnected and unkown.

TommasoPino, 2023-03-28:
Hi Pigi-102, the configuration sofar_hyd3k-6k-es.yaml exposes the 77
entities because the configuration defines them. The entities remain
disconnected end unknown because the register addresses defined in the
configuration file are not correct for your inverter.

The sofar_g3hyd.yaml is the configuration for a 3 Phase inverter (R, T, S
are the name of each phase) but it is also valid for the 1 Phase inverter.
The reason behind that is due to programming efficiency. There is no reason
to write twice the firmware code for the same controller computer installed
for both inverter families. Indeed, the registers for the first phase are
the same for both inverters.

I have no time at this moment to work on it and clean the sofar_g3hyd.yaml
from the unnecessary entities creating a sofar_g1hyd.yaml configuration.
But I ensure you that the configuration identified is correct for your
inverter too. I also tested the writing features and it works perfectly.

Pigi-102, 2023-03-29:
Hello, I started to create a sofar_g1hyd.yaml by removing lot ( if not all ) of "T" "R" and "S" entities. There are some that return a value also if are labeled with those letter. 
I would like to know if there is some paper that explain the meaning o some of this entities.
I cannot find, for example a value that tell me how much power my PV are sending to the grid or others. 
I know it's a little off topic but maybe you can, as side note, help me in this.

here the file:
[sofar_g1hyd.yaml.gz](https://github.com/StephanJoubert/home_assistant_solarman/files/11101996/sofar_g1hyd.yaml.gz)

kavejo, 2023-03-29:
@Pigi-102, you have to open a ticket with Zucchetti and ask them to send you the registers definition for your inverter model. That will tell exactly what you can get, at what address to read each entry, etc.

TommasoPino, 2023-03-29:
> Hello, I started to create a sofar_g1hyd.yaml by removing lot ( if not all ) of "T" "R" and "S" entities. There are some that return a value also if are labeled with those letter. I would like to [...]

Hi @Pigi-102, I will share with you the file that i found surfing the web. Use it carrefully because also the installer writing addresses are mapped. Please consider to use only the EndUser registers. The SpreadSheet is in Chinese, on the third sheet there is the entire map.

[Link](https://docs.google.com/spreadsheets/d/1aZc8tFrxvYOPeL6Y13sDsQHNn7XQ4pqb/edit?usp=share_link&ouid=105839117791356844874&rtpof=true&sd=true)

In order to correctly map the 1Ph respect to 3Ph you do not have to remove all reference at T, R ans S phase because in that way the information you want will be lost. Consider instead to remove all reference to T and S phases and rename the R information removing the _R. 
This will produece a configuration file close to the final one, the only thing to add will be to remove the addresses ask to the modbus but it is something that could be done later.

Hope to give you the information searched. I hope also to find the time needed to share with all of you the inverter control capability I told you before. This just a screen :)

[image]
[image]

Pigi-102, 2023-03-30:
Thanks @TommasoPino .
I will look into it and hope to be able to do something.
My approach has been to remove lot of Reference with "Unkown" as result and the three "T" "R" and "S" if the value returned was zero or unknown and there was a "Total" with a value insted.
For example, ActivePower_Load_[RST] IIRC where at 0, but ActivePower_Load_Total had a (valid) value, confirmed with the value read from the app. In that case I removed the three and left only the Total.
Current_Output_R instead had a value > 0 and I decided to leave him in yaml. Same goes for Current_PCC_R, Voltage_Phase_R and eventually other with a value.

Just to avoid working on an 3ph outdated, you advise to use the latest from main branch or one from a determined branch ?

paqpal, 2023-05-02:
Hi,
I have a ZCS hyd6000 and wifi stick LSW3_15_FFFF_1.0.78
A year ago I wrote a custom yaml trying to map as much as I can

Here is the file i posted here on github
#https://github.com/StephanJoubert/home_assistant_solarman/issues/45#issuecomment-1121059303

I think it was embedded in this project.
It worked perfectly until 26 April 2023 at 10:33.

Right now logs are:

```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solarman/__init__.py", line 35, in update_listener
    hass.data[DOMAIN][entry.entry_id].config(entry)
KeyError: 'solarman'
```

and 

`
Updating solarman sensor took longer than the scheduled update interval 0:00:30`

`Update of sensor.solarman_pv_instant_generated_pw is taking over 10 seconds`

last 

`Querying registers [512 - 597] failed, aborting.`

That day I installed new Ring Intercom, nothing more.
My inverter is under firewall only on local lan.
I tried my custom yaml and that one provided by addon.

picardi21, 2023-07-27:
> lema per permettere a tutti di risolvere questo problema... grazie alla condivisione delle informazioni tra te e ZCS

ma dove trovi il file

> le definizioni dell'inverter ho visto che **sofar_g3hyd.yaml** corrisp

but can you see the instantaneous production of photovoltaics??

picardi21, 2023-07-27:
> i, sono stato in grado di integrare l'inverter in HA utilizzando l'ultima versione dell'estensione Solarman.
> Devi riscaricare l'estensione utilizzando l'ultima versione disponibile. È necessario abilitare "versioni beta" e quindi selezionare "principale".
> Ha funzionato per me.
> Spero che questo ti aiuti

ma dove si trova il file da aggiungere nella cartella?

giuliovialetto, 2024-03-07:
> Hi @Pigi-102, I will share with you the file that i found surfing the web. Use it carrefully because also the installer writing addresses are mapped. Please consider to use only the EndUser [...]

Hi @TommasoPino is this available to manage inverter from HomeAssistant?

TommasoPino, 2024-03-12:
> Hi @TommasoPino is this available to manage inverter from HomeAssistant?

Hello @giuliovialetto, everyone,

I finally had time to fork this repo and apply what I made. The following is the repo where at the moment I had control of "Inverter Storage Mode" and recently I added also a button to trigger the scan on the IV curve to find a better working point. I need to add this because the secondary PV string did not work properly (now the problem seems to be solved with an automation that presses the button every 30 minutes)

[https://github.com/TommasoPino/home_assistant_solarman_w_control](https://github.com/TommasoPino/home_assistant_solarman_w_control)

I hope someone else could help me to continue the activity.

Have a good time everyone,
Tommaso

giuliovialetto, 2024-03-12:
> Hello @giuliovialetto, everyone,
>
> I finally had time to fork this repo and apply what I made. The following is the repo where at the moment I had control of "Inverter Storage Mode" and recently I [...]

Thank you Tommaso!

TommasoPino, 2025-05-30:
after a while and after several investigation, I found another integration that collect all the data but also allow the inverter control

https://github.com/wills106/homeassistant-solax-modbus

thanks so much to the author of home_assistant_solarman custom integration, but I think I am moving to that new one