StephanJoubert/home_assistant_solarman #162: Totals are not available in energy grid for sofar_hyd3k-6k.yaml

murraythegoz, 2023-01-01:
I just installed the integration but although I can see the single sensors, the "totals" are not made available on Energy config when setting consumption and return.
Only "today" params are available

tried to change the state class to "total_increasing" instead of "total" and reinstanced the component and integration, but with no success.

Any hint?

mrwolfjunior, 2023-01-10:
> I just installed the integration but although I can see the single sensors, the "totals" are not made available on Energy config when setting consumption and return. Only "today" params are [...]

Hi,

I'm not able to test it right now, but if you add this template to your configuration.yaml, should fix the issue:

```
template:
  - sensor:
      - name: "Prelievo Totale"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.zcs_total_grid_consumption') | float (0) }}"
        availability: "{{ states('sensor.zcs_total_grid_consumption')|is_number }}"
        state_class: total_increasing
        device_class: energy
        icon: mdi:transmission-tower
      - name: "Immissione Totale"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.zcs_total_grid_return') | float(0) }}"
        availability: "{{ states('sensor.zcs_total_grid_return')|is_number }}"
        state_class: total_increasing
        device_class: energy
        icon: mdi:transmission-tower-export
      - name: "Scarica Totale"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.zcs_battery_total_energy_dischaged') | float(0) }}"
        availability: "{{ states('sensor.zcs_battery_total_energy_dischaged')|is_number }}"
        state_class: total_increasing
        device_class: energy
        icon: mdi:battery-low
      - name: "Carica Totale"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.zcs_battery_total_energy_charged') | float(0) }}"
        availability: "{{ states('sensor.zcs_battery_total_energy_charged')|is_number }}"
        state_class: total_increasing
        device_class: energy
        icon: mdi:battery-high
      - name: "Produzione Totale"
        unit_of_measurement: "kWh"
        state: "{{ states('sensor.zcs_total_production') | float(0) }}"
        availability: "{{ states('sensor.zcs_total_production')|is_number }}"
        state_class: total_increasing
        device_class: energy
        icon: mdi:weather-sunny

```

Make sure to change the sensor name inside the "state" and "availability" attributes... Let me know!

murraythegoz, 2023-01-10:
Hi,

Just tested (yes, with proper sensor customization), and still just the "today" are shown. 
totals are not selectable in energy config, but works as independent sensors.

mrwolfjunior, 2023-01-10:
> Hi,
> 
> Just tested (yes, with proper sensor customization), and still just the "today" are shown. totals are not selectable in energy config, but works as independent sensors.

Yes, this is correct; if you have inserted the template above and restarted home assistant, you should see five additional sensors (the id of the sensor is generated based on the "name" attribute):

- sensor.carica_totale
- sensor.immissione_totale
- sensor.prelievo_totale
- sensor.produzione_totale
- sensor.scarica_totale

So, in the energy dashboard, you must select the sensors above instead of the integration one. Like so:

[image]

murraythegoz, 2023-01-11:
The templates worked well.

thank you

picardi21, 2023-07-27:
Hi where do I find the sofar_hyd3k-6k.yaml file to add in the "inverter_definitions" folder