StephanJoubert/home_assistant_solarman #169: Recommendation for SN23...

proasnet, 2023-01-08:
Hello, I tried your integration on system of my friend. I found, that him solarmans has SN with 23... and integration not working. I know, that must be with 17...
Is any way, how read data from inverters to HA?
Thank you very much.

blackscreener, 2023-01-08:
Wrong registers?

blackscreener, 2023-01-08:
Wrong registers?

svenwal, 2023-01-11:
I do have a hyd 20ktl-3ph and a SN23xxxx logger and for me solar_g3hyd has been the one being able to get all data out of the box

PatrykCzarnota, 2023-01-21:
I also have sofar solar (Firmware version LSW3_15_270A_1.50) with Device serial number 23xxxxxxxx and it's not working with this intergration.

ChrisPank, 2023-02-08:
> I do have a hyd 20ktl-3ph and a SN23xxxx logger and for me solar_g3hyd has been the one being able to get all data out of the box

Can confirm this, solar_g3hyd works for HYD 10ktl-3ph with SN23xxxx logger (Firmware LSW3_15_270A_1.50).

PatrykCzarnota, 2023-02-08:
> > I do have a hyd 20ktl-3ph and a SN23xxxx logger and for me solar_g3hyd has been the one being able to get all data out of the box
> 
> Can confirm this, solar_g3hyd works for HYD 10ktl-3ph with SN23xxxx logger (Firmware LSW3_15_270A_1.50).

@ChrisPank Wow, could you share your settings? Maybe I do something wrong.

Nikki1905, 2023-02-14:
Hi there, I have the Firmware version _LSW3_15_270A_1.53_ with SOFAR HYD 10KTL-3PH-Hybrid and die logger SN239XXXXXX and the solar_g3hyd will not work :/

ChrisPank, 2023-02-14:
> Hi there, I have the Firmware version _LSW3_15_270A_1.53_ with SOFAR HYD 10KTL-3PH-Hybrid and die logger SN239XXXXXX and the solar_g3hyd will not work :/

Are you able to get the data on any other way? There are some lsw-3 stick logger "reading-scripts" written in python or GO here on github. If they work, its propably not a stick-firmware related problem. Maybe they changed sth on config_hide.html with latest firmware but you could check there if is a tcp server running.

Nikki1905, 2023-02-22:
> Hi there, I have the Firmware version _LSW3_15_270A_1.53_ with SOFAR HYD 10KTL-3PH-Hybrid and die logger SN239XXXXXX and the solar_g3hyd will not work :/

Yesterday I updatet the Solarman HACS integration to v1.4.3 and now I see information/data from the inverter. 

I can confirm that the config **work**!
Firmware version LSW3_15_270A_1.53
SOFAR HYD 10KTL-3PH-Hybrid
logger SN239XXXXXX
solar_g3hyd

fsilberstein, 2023-02-23:
I also spent a bit of time finding the right settings and tried several configurations. 
But in the end, I can also confirm that the following works for me:

Firmware version LSW3_15_270A_1.53
SOFAR HYD 3000-EP
logger SN236XXXXXX
solar_g3hyd

Thanks again for the great work @StephanJoubert

cvandesande, 2023-07-30:
Thank you @fsilberstein your comment helped me resolve my issue. This integration is working wonderfully!

Firmware version LSW3_15_270A_1.65
SOFAR HYD 6000-EP (6KTL-EP)
logger SN239XXXXXX
solar_g3hyd

closed 2024-03-13 by proasnet