StephanJoubert/home_assistant_solarman #181: Deye inverters in parallel

zkrige, 2023-01-19:
I have 2 inverters and 2 data loggers. Is there a way to consolidate the data from multiple loggers ?

moreje, 2023-02-10:
you can build your own sensors templates that add them together

Klagio, 2023-02-10:
> you can build your own sensors templates that add them together

Sure but the production data is already the sum

Klagio, 2023-02-10:
> I have 2 x 8kw Deye inverters and 2 data loggers. Is there a way to consolidate the data from multiple loggers ?

you can get both data in Home Assistant? I am able to get only from one instance, the second I receive all UNKNOWN values

francis91joseph, 2023-04-20:
@Klagio were you able to fix this issue? I am facing the same.

branteforest, 2023-05-15:
hi there. i have the same problem. does anyone have a different solution?

francis91joseph, 2023-05-17:
> hi there. i have the same problem. does anyone have a different solution?

Hi @branteforest, the problem can be solved by using a custom yaml file that contains updated registers for DEYE 12KW inverters.

Check this

https://github.com/StephanJoubert/home_assistant_solarman/issues/88

d3nho, 2023-06-27:
Hi, did anyone manage to find a solution for having multiple inverters in parallel?

Smiley-Ret, 2023-09-24:
I want to add a second Deye inverter to Homeassistant,  that is functioning in parallel with the first, but ADD / configuring the second inverter on Solarman does not work. All the entities are unavailable in the second configuration.

Physically the two inverters and dataloggers are active on the Solarman app, and show current data.

How can I configure both DEYE 5kW inverters into Homeassistant under the community's add on for Solarman?

PS: Both have fixed IPs

[image]
[image]
[image]
[image]

deuko, 2023-10-21:
Bump, I have the same issue.

I have running 2x 12K-SG04LP3-EU in parallel (one is master with mb address 01, second one is slave with mb address 02). Both are also shown in the Solarman App, both loggers are working.
Both have a fixed IP address.
I can reach both via the web UI.

But also only the first one works for me in Home Assistant.
Both have the same configuration (deye_sg04lp3.yaml) when adding, I tried different mb addresses for the second one, no luck.

Smiley-Ret, 2023-10-23:
Hi, I tried all I could think of, but in the end, I deleted the solarman integration , restarted Homeassistant and loaded solarman again.

Smiley-Ret, 2023-10-23:
After that, I was able to load both and works fine for the the period til now.

Smiley-Ret, 2023-10-23:
[image]

deuko, 2023-10-23:
@Smiley-Ret Nice, do you mind sharing what configurations you have on your inverters and the solarman integration? Like the modbus master or slave, and which modbus address you gave them.

Smiley-Ret, 2023-10-23:
I used the power flow card from slipx06, https://slipx06.github.io/sunsynk-power-flow-card/index.html.
I spent some time to put the Solarman entities into the card and just read the modbus through the HACS add on. I am still learning, but will tackle direct modbus settings later on. I also used a card for each inverter.

[image]

deuko, 2023-10-23:
Ok, removing and re-adding everything did not work in my case, but I think found the underlying issue, at least in my case, don't know if its the same for the others.

I checked the logs and found that for the second inverter, no valid battery temperature could be read:
[image]

After this, I checked the readings on my inverter itself, and indeed, the second inverter (slave) seems to not get correct battery readings from the BMS. (I have Pytes Batteries with a Pytes Battery HUB, the HUB is only connected to the first (master) inverter)

This is the battery info on the master - correct reading of 24.2°C:
[image]
This is the battery info on the slave - incorrect / no values:
[image]

It shows -100°C on the display, the interesting thing is however, if I query both inverters using only the PySolarmanv5 lib:
`inverterSo = PySolarmanV5("192.168.31.126", 2782677988)
inverterSw = PySolarmanV5("192.168.31.33", 2793644608)

print(inverterSo.read_holding_registers(register_addr=586, quantity=1))
print(inverterSw.read_holding_registers(register_addr=586, quantity=1))`

It gave me the following output:
[1242]
[0]

The first line is correct, the value needs to be offset by 1000, as defined in the yaml -> this means 24.2
The second line of the slave inverter obviously produces a wrong value.

IMO there should be an option to "allow" false values, or not to discard the whole dataset, so that at least the other values are shown.

@StephanJoubert whats your take on this? Maybe you have an idea? Appreciated.

Smiley-Ret, 2023-10-23:
I have the battery temp problem on both inverters:
I did not debug very far on that issue, still one of many little bugs to resolve, but just needed to mention that with that temp, both my inverters still work on home assistant. I have seen screen shots from other people with the -100°C, so it seems to be prevalent. 

[image]

Smiley-Ret, 2023-10-23:
Here is my Yaml for the master inverter card:
I have many custom sensors to produce one card in the end that shows the two Inverter as if one. See next message.

type: custom:sunsynk-power-flow-card
cardstyle: full
panel_mode: false
large_font: true
show_solar: true
inverter:
  modern: true
  colour: '#FFFFFF'
  autarky: none
battery:
  energy: 15000
  shutdown_soc: 20
  invert_power: false
  colour: '#0066FF'
  show_daily: true
  animation_speed: 4
  max_power: 4800
  full_capacity: 80
  empty_capacity: 20
solar:
  colour: orange
  show_daily: false
  mppts: 1
  animation_speed: 8
  max_power: 5000
  pv1_name: Garage
  pv2_name: House
  pv3_name: none
  pv4_name: none
load:
  colour: '#66CCFF'
  show_daily: false
  show_daily_aux: false
  show_aux: true
  invert_aux: false
  aux_colour: '#33CC33'
  aux_off_colour: red
  animation_speed: 8
  max_power: 7200
grid:
  colour: '#FF3300'
  show_daily_buy: false
  show_daily_sell: false
  no_grid_colour: '#a40013'
  show_nonessential: false
  invert_grid: false
  nonessential_name: Non Essential
  nonessential_icon: oven
  load1_icon: boiler
  load2_icon: mdi:ev-station
  animation_speed: 8
  max_power: 5000
entities:
  aux_connected_status: none
  aux_load1: sensor.stove_power
  aux_load2: sensor.geyser_power
  aux_power_166: sensor.half_geyser_and_stove_power
  battery_current_191: sensor.solarman_master_battery_current
  battery_power_190: sensor.solarman_master_battery_power
  battery_soc_184: sensor.solarman_master_battery_soc
  battery_temp_182: none
  battery_voltage_183: sensor.solarman_master_battery_voltage
  day_aux_energy: sensor.daily_geyser_plus_stove_energy
  day_battery_charge_70: sensor.solarman_master_daily_battery_charge
  day_battery_discharge_71: sensor.solarman_master_daily_battery_discharge
  day_grid_export_77: none
  day_grid_import_76: none
  day_load_energy_84: sensor.daily_household_consumption
  day_pv_energy_108: sensor.solarman_master_daily_production
  dc_transformer_temp_90: sensor.solarman_master_dc_temperature
  grid_connected_status_194: sensor.solarman_master_grid_connected_status
  grid_ct_power_172: none
  grid_power_169: none
  inverter_current_164: sensor.solarman_master_current_l1
  inverter_power_175: sensor.solarman_master_inverter_l1_power
  inverter_status_59: sensor.solarman_master_running_status
  inverter_voltage_154: sensor.solarman_master_load_voltage
  load_frequency_192: sensor.solarman_master_load_frequency
  pv_total: sensor.solarman_slave_pv1_power
  pv2_current_110: sensor.solarman_master_pv1_current
  pv1_power_186: sensor.solarman_slave_pv1_power
  pv1_voltage_109: sensor.solarman_master_pv1_voltage
  pv2_current_112: sensor.solarman_master_pv2_current
  pv2_power_187: sensor.solarman_master_pv2_power
  pv2_voltage_111: sensor.solarman_master_pv2_voltage
  radiator_temp_91: sensor.solarman_master_ac_temperature
  essential_power: sensor.half_essential_load_power

Smiley-Ret, 2023-10-23:
This is my combined card, and my seperate UPS on a Luxpower inverter:

[image]

bmatti996, 2024-08-13:
hi, I have the same problem, I have 2 6kw deye inverters and I can't see the slave, I only see the master, any solution?
[image]

PV-Joe, 2024-08-13:
> hi, I have the same problem, I have 2 6kw deye inverters and I can't see the slave, I only see the master, any solution? ![slave]

I suggest you try the updated Solarman from here:
https://github.com/davidrapan/ha-solarman