StephanJoubert/home_assistant_solarman #187: Connection with inverter (Sofar non-hybrid) lost after upgrade to 1.4.2

JHurk, 2023-01-21:
Have been using the integration for a few months and has always worked flawless.
After updating to 1.4.2 the integration doesn't get any connection to the inverter.
I do not see anything in de log-files (or I am looking in the wrong place).

`Logger: homeassistant.helpers.entity
Source: helpers/entity.py:538 
First occurred: 16:13:52 (49 occurrences) 
Last logged: 18:16:52

Update of sensor.zonnepanelen_pv1_power is taking over 10 seconds`

When I downgrade to version 1.4.1 everything works fine again and the connection gets established instantly after te restart.
I tried upgrading three times, everytime the same issue occurs.

I have the following configuration;
`sensor:
  - platform: solarman
    name: Zonnepanelen 
    inverter_host: 192.168.X.XXX (IP-address)
    inverter_port: 8899
    inverter_serial: XXXXXXXXX (the correct inverter_serial) 
    inverter_mb_slaveid: 1
    lookup_file: sofar_lsw3.yaml`
This setup works on 1.4.1 but not on 1.4.2.
I see there are new/extra lookup files in 1.4.2, but the one I use is still in 1.4.2 and selected.

I understand the information is limited, but if needed I can provide extra information.

heebo1974, 2023-01-21:
Same here..

```
This error originated from a custom integration.

Logger: homeassistant.helpers.entity
Source: custom_components/solarman/solarman.py:234
Integration: Solarman (documentation, issues)
First occurred: January 20, 2023 at 07:16:46 (31395 occurrences)
Last logged: 22:57:52

Update for sensor.solar_sofar_daily_production fails
Update for sensor.solar_sofar_total_production fails
Update for sensor.solar_sofar_total_generation_time fails
Update for sensor.solar_sofar_today_generation_time fails
Update for sensor.solar_sofar_output_active_power fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 736, in async_device_update
    raise exc
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solarman/sensor.py", line 140, in update
    self.inverter.update()
  File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 192, in wrapper
    result = method(*args, **kwargs)
  File "/config/custom_components/solarman/solarman.py", line 175, in update
    self.get_statistics()
  File "/config/custom_components/solarman/solarman.py", line 234, in get_statistics
    sock.close()
UnboundLocalError: local variable 'sock' referenced before assignment
```

alessandroo23, 2023-01-22:
same problem after update.

MaciekMordal, 2023-01-22:
same here
sofar_lsw3.yaml

MaciekMordal, 2023-01-22:
[custom_components.solarman.solarman] Querying registers [261 - 276] failed, aborting.

WVoo, 2023-01-22:
Also after update to 1.4.2 Solarman can't acquire data. Reverted to 1.4.1 - and works again

lukasniemeier, 2023-01-30:
Let me check what I've messed up when distilling the changes from my local branches :see_no_evil:

lukasniemeier, 2023-01-30:
Can you try the change in https://github.com/StephanJoubert/home_assistant_solarman/pull/194? Also if you can post a full log on `debug` level of the component (see [here](https://www.home-assistant.io/integrations/logger/)) active would be helpful :slightly_smiling_face:

JHurk, 2023-01-31:
Not able to do this myself as we just went on holiday for the next 10 days. So hopefully someone else can test this? Thanks for the effort to try and make this work again.

dsgepers, 2023-01-31:
```
2023-01-31 08:46:37.760 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Starting to query for [2] ranges...
2023-01-31 08:46:37.762 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [0 - 39]...
2023-01-31 08:46:37.762 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030000002845d4f815
2023-01-31 08:46:38.150 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a56300101500a66301647d0201828831014c050000da3ea7620103500002000000000000000000000bf700060b0f00000002000000020000138808b6000f0000000000000000000012900000145600000007001000200ddb0bf70b0f003c000000010000029b2ee02ee0000c96cab115
2023-01-31 08:46:38.150 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [0 - 39] succeeded
2023-01-31 08:46:38.151 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276]...
2023-01-31 08:46:38.151 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:39.270 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500a76301647d0201848831014d050000da3ea7620500da15
2023-01-31 08:46:39.270 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:39.270 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:39.270 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [5] retry attempts left
2023-01-31 08:46:39.270 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:40.390 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500a86301647d0201858831014e050000da3ea7620500dd15
2023-01-31 08:46:40.390 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:40.390 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:40.390 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [4] retry attempts left
2023-01-31 08:46:40.390 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:41.510 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500a96301647d0201868831014f050000da3ea7620500e015
2023-01-31 08:46:41.510 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:41.510 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:41.510 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [3] retry attempts left
2023-01-31 08:46:41.510 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:42.630 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500aa6301647d02018788310150050000da3ea7620500e315
2023-01-31 08:46:42.630 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:42.630 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:42.630 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [2] retry attempts left
2023-01-31 08:46:42.630 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:43.750 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500ab6301647d02018888310151050000da3ea7620500e615
2023-01-31 08:46:43.750 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:43.750 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:43.750 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [1] retry attempts left
2023-01-31 08:46:43.750 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 08:46:44.870 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] a51000101500ac6301647d02018988310153050000da3ea7620500ea15
2023-01-31 08:46:44.870 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 08:46:44.870 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:46:44.870 DEBUG (SyncWorker_10) [custom_components.solarman.solarman] Querying [261 - 276] failed, [0] retry attempts left
2023-01-31 08:46:44.870 WARNING (SyncWorker_10) [custom_components.solarman.solarman] Querying registers [261 - 276] failed, aborting.
2023-01-31 08:47:37.762 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] Starting to query for [2] ranges...
2023-01-31 08:47:37.765 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] Querying [0 - 39]...
2023-01-31 08:47:37.765 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030000002845d4f815
[... cut ...]
2023-01-31 08:47:44.873 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 08:47:44.873 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] Querying [261 - 276] failed, [0] retry attempts left
2023-01-31 08:47:44.873 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying registers [261 - 276] failed, aborting.
```

@lukasniemeier 

Also quickly additionally tagging @steve-cox for seeing "invalid modbus crc" message in debug.

lukasniemeier, 2023-01-31:
@dsgepers thanks for the debug output, it shows that on querying the second register range the inverter/data-logger returns garbage data for the modbus frame (at least I cannot make anything out of `0x0500`).

Assuming that the problem originates from re-using the socket, I have adjusted https://github.com/StephanJoubert/home_assistant_solarman/pull/194 to the old behavior: a fresh connection for each request range. I did that change originally as my inverter didn't like having a new socket opened for every range, but let's test first if this resolve the problem here. Mind that I am not able to test this until later, but feel free to give it a try

dsgepers, 2023-01-31:
For when you have a moment to test, here's my log with that last change:

```
2023-01-31 13:43:11.747 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Starting to query for [2] ranges...
2023-01-31 13:43:11.749 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [0 - 39]...
2023-01-31 13:43:11.749 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030000002845d4f815
2023-01-31 13:43:12.094 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a56300101500df6301647d020157cd31017e4a0000863fa7620103500002000000000000000000000b8700570e9400460019001a002b0000138a08f500bd0000000000000000000012910000145b00aa012f001800260e870b870e94003c000000010000029b2ee02ee0000c67946715
2023-01-31 13:43:12.095 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [0 - 39] succeeded
2023-01-31 13:43:12.096 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276]...
2023-01-31 13:43:12.097 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:13.216 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e06301647d020158cd31017f4a0000863fa76205005015
2023-01-31 13:43:13.216 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:13.216 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:13.216 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [5] retry attempts left
2023-01-31 13:43:13.216 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:14.338 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e16301647d020159cd3101804a0000863fa76205005315
2023-01-31 13:43:14.338 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:14.338 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:14.338 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [4] retry attempts left
2023-01-31 13:43:14.338 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:15.460 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e26301647d02015acd3101814a0000863fa76205005615
2023-01-31 13:43:15.460 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:15.460 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:15.460 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [3] retry attempts left
2023-01-31 13:43:15.460 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:16.581 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e36301647d02015bcd3101834a0000863fa76205005a15
2023-01-31 13:43:16.582 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:16.582 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:16.582 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [2] retry attempts left
2023-01-31 13:43:16.582 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:17.713 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e46301647d02015ccd3101844a0000863fa76205005d15
2023-01-31 13:43:17.713 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:17.714 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:17.714 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [1] retry attempts left
2023-01-31 13:43:17.714 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500006301647d02000000000000000000000000000001030105001055fb1d15
2023-01-31 13:43:18.825 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500e56301647d02015dcd3101854a0000863fa76205006015
2023-01-31 13:43:18.825 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-01-31 13:43:18.825 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:43:18.825 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [261 - 276] failed, [0] retry attempts left
2023-01-31 13:43:18.825 WARNING (SyncWorker_8) [custom_components.solarman.solarman] Querying registers [261 - 276] failed, aborting.
2023-01-31 13:44:11.748 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] Starting to query for [2] ranges...
2023-01-31 13:44:11.750 DEBUG (SyncWorker_11) [custom_components.solarman.solarman] Querying [0 - 39]...
[... cut ...]
2023-01-31 13:46:18.289 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] unexpected v5 control code
2023-01-31 13:46:18.289 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
2023-01-31 13:46:18.289 DEBUG (SyncWorker_5) [custom_components.solarman.solarman] Querying [261 - 276] failed, [0] retry attempts left
2023-01-31 13:46:18.289 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying registers [261 - 276] failed, aborting.
```

lukasniemeier, 2023-01-31:
@dsgepers thanks for testing, now seeing the logs of version 1.4.1 on https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1409911032 (missed the edit this morning on mobile) I have reason to believe that your problem actually is connected to https://github.com/StephanJoubert/home_assistant_solarman/pull/125 (a different change that landed with 1.4.2 - see https://github.com/StephanJoubert/home_assistant_solarman/compare/v1.4.1...v1.4.2).

In both your 1.4.1 and 1.4.2 logs we see that your inverter/data-logger response with the modbus frame `0x0500` for the registers `261 - 276`. See below for a 1.4.1 and a 1.4.2 response from the logs you shared above, I have highlighted the modbus frame.

```
a510001015000b6301647d0201198f3101ec0b0000e83ea762 - 0500 - 8d15
a51000101500a76301647d0201848831014d050000da3ea762 - 0500 - da15
```

Before https://github.com/StephanJoubert/home_assistant_solarman/pull/125 this (apparently) malformed message, was simply ignored (no data extracted for registers `261 - 276`).

@dsgepers would you mind testing to apply https://github.com/StephanJoubert/home_assistant_solarman/pull/196 to v1.4.2 ?
@heebo1974 are you also using `sofar_lsw3` ?

heebo1974, 2023-01-31:
> @dsgepers thanks for testing, now seeing the logs of version 1.4.1 on [#187 (comment)](https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1409911032) (missed the edit this morning on mobile) I have reason to believe that your problem actually is connected to #125 (a different change that landed with 1.4.2 - see [v1.4.1...v1.4.2](https://github.com/StephanJoubert/home_assistant_solarman/compare/v1.4.1...v1.4.2)).
> 
> In both your 1.4.1 and 1.4.2 logs we see that your inverter/data-logger response with the modbus frame `0x0500` for the registers `261 - 276`. See below for a 1.4.1 and a 1.4.2 response from the logs you shared above, I have highlighted the modbus frame.
> 
> ```
> a510001015000b6301647d0201198f3101ec0b0000e83ea762 - 0500 - 8d15
> a51000101500a76301647d0201848831014d050000da3ea762 - 0500 - da15
> ```
> 
> Before #125 this (apparently) malformed message, was simply ignored (no data extracted for registers `261 - 276`).
> 
> @dsgepers would you mind testing to apply #196 to v1.4.2 ? @heebo1974 are you also using `sofar_lsw3` ?

Yep, sofar lsw3

steve-cox, 2023-01-31:
@lukasniemeier Looking at the packets you have highlighted the modbus frame in, it's clear something is wrong....

We can use [https://pysolarmanv5.readthedocs.io/en/latest/solarmanv5_protocol.html](https://pysolarmanv5.readthedocs.io/en/latest/solarmanv5_protocol.html) and [http://bends.se/?page=anteckningar/automation/modbus/frame-format](http://bends.se/?page=anteckningar/automation/modbus/frame-format) to understand the frames.  Picking apart the working one, the entire modbus frame is long highlighted section:
`a5` `6300` `1015` `00f5` `6301647d` `02` `01` `09ce3101` `314b0000` `893fa762` **`0103500002000000000000000000000bd900350e9200270010000e00180000138a08e2006b0000000000000000000012910000145b00ac0132001800260e840bd90e92003c000000010000029b2ee02ee0000cbed5`** `53` `15`

...and that modbus frame is:
`01` `03` `50` `0002000...<data>...e0000c` `bed5`

0x01 = unit address, 0x03 = function code (ie, repeating what function we asked it to perform - 'read holding registers'),
0x50 = number of bytes, followed by those data bytes,
0xbed5 = checksum.

All good.  Now lets pick apart a bad frame, modbus frame highlighted:

`a5` `6300` `1015` `000b` `6301647d` `02` `01` `198f3101` `ec0b0000` `e83ea762` **`0500`** `8d` `15`

....and that modbus frame is.... completely invalid - it's only 2 bytes long.  We need a unit address, the function code, the number of bytes, the data itself, and a checksum.  Even if we could get a response with zero bytes (not sure that's valid), it would have to be a minimum of 5 bytes long.  Oddly enough the encapsulating solarmanv5 frame is correct, I'm guessing the data logger just wrapped up whatever packet it received from the inverter.

Any way, the frames are definitely invalid and I think the code is correctly detecting that.... although we could also check for a minimum modbus frame length, just in case the 'checksum' accidentally calculates.

So... maybe the inverter doesn't actually have anything at these addresses?

Unless my eyes are deceiving me, I have noticed that the sofar_lsw3.yaml isn't actually do anything with the registers at 261-276 [0x0105-0x0114] anyway.... maybe simply removing them resolves this issue?

@dsgepers @JHurk: I had to change the file type from .yaml to .txt in order to attach, but does this custom_parameters.yaml fix the problem?
[custom_parameters.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/10549983/custom_parameters.txt)

lukasniemeier, 2023-01-31:
> Any way, the frames are definitely invalid and I think the code is correctly detecting that.... 

Yes, exactly what I was describing in https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1410900720

> So... maybe the inverter doesn't actually have anything at these addresses?

Yes, that's why I was proposing https://github.com/StephanJoubert/home_assistant_solarman/pull/196 as a potential solution for the problem described in this issue.

steve-cox, 2023-01-31:
> > Any way, the frames are definitely invalid and I think the code is correctly detecting that....
> 
> Yes, exactly what I was describing in [#187 (comment)](https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1410900720)
> 
> > So... maybe the inverter doesn't actually have anything at these addresses?
> 
> Yes, that's why I was proposing #196 as a potential solution for the problem described in this issue.

Ahh, my mistake.  I misunderstood "(apparently) malformed message" to mean that you didn't think it was an invalid packet :)

It did occur to me that it might be nice to be able to accept the responses of just the ranges get results, rather than dropping the whole lot.... but then again, it may make other problems harder to see....

lukasniemeier, 2023-01-31:
> although we could also check for a minimum modbus frame length, just in case the 'checksum' accidentally calculates.

If https://github.com/StephanJoubert/home_assistant_solarman/pull/196 proves to be effective, I am +1 on your proposal to actually make the validation logic even stricter than right now, i.e. extending it to some basic modbus-frame-validation. This way one catches "bugs" in the parameter file faster.

dsgepers, 2023-02-02:
@steve-cox Yes! This (https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1411000469) works for me 100%. This is on a fresh install of 1.4.2 without any of the other changes.

lukasniemeier, 2023-02-03:
based on your testing @dsgepers I guess https://github.com/StephanJoubert/home_assistant_solarman/pull/196 is the way to go then for a projected `1.4.3`

dsgepers, 2023-02-03:
I would agree, thank you and @steve-cox for your effort. I do hope we can put a 1.4.3. together in a reasonable timeframe as the current version does seem to generate a lot of issues. @StephanJoubert ?

JHurk, 2023-02-08:
@lukasniemeier @steve-cox @dsgepers 
Just got back from traveling, so got al little bit of catching up to do.
Am I correct in my assumption that the cause of this issue has been found, a fix is projected and we are now waiting for a new release (1.4.3) in which the bug fix can be implemented?
Can I help in any way (without the need to adjust code because I lack knowledge and think I will break more then I can fix :))

lukasniemeier, 2023-02-11:
@JHurk in the end it seems like we are looking at a couple of different challenges, maybe you can try activating debug logs for the solarman component (see [here](https://www.home-assistant.io/integrations/logger/)) - with that we may be able to see if your issue is related

JHurk, 2023-02-11:
`2023-02-11 11:43:51.835 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-02-11 11:44:02.549 DEBUG (MainThread) [custom_components.solarman] __init__.py:async_setup_entry({'entry_id': '0457239070688fdd98c20aecf1a418b3', 'version': 1, 'domain': 'solarman', 'title': '192.168.1.73', 'data': {'name': 'Zonnepanelen', 'inverter_host': '192.168.1.73', 'inverter_serial': 520018464, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'sofar_lsw3.yaml'}, 'options': {'name': 'Zonnepanelen', 'inverter_host': '192.168.1.73', 'inverter_serial': 520018464, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'sofar_lsw3.yaml'}, 'pref_disable_new_entities': False, 'pref_disable_polling': False, 'source': 'user', 'unique_id': None, 'disabled_by': None})
2023-02-11 11:44:02.555 WARNING (MainThread) [homeassistant.helpers.frame] Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for solarman using this method at custom_components/solarman/__init__.py, line 18: hass.config_entries.async_setup_platforms(entry, PLATFORMS)
2023-02-11 11:44:02.645 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_entry: {'name': 'Zonnepanelen', 'inverter_host': '192.168.1.73', 'inverter_serial': 520018464, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'sofar_lsw3.yaml'}
2023-02-11 11:44:02.645 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:async_setup_platform: {'name': 'Zonnepanelen', 'inverter_host': '192.168.1.73', 'inverter_serial': 520018464, 'inverter_port': 8899, 'inverter_mb_slaveid': 1, 'lookup_file': 'sofar_lsw3.yaml'}
2023-02-11 11:44:02.807 DEBUG (MainThread) [custom_components.solarman.sensor] sensor.py:_do_setup_platform: async_add_entities
2023-02-11 11:44:02.807 DEBUG (MainThread) [custom_components.solarman.sensor] [<Entity Zonnepanelen PV1 Power: None>, <Entity Zonnepanelen PV2 Power: None>, <Entity Zonnepanelen PV1 Voltage: None>, <Entity Zonnepanelen PV2 Voltage: None>, <Entity Zonnepanelen PV1 Current: None>, <Entity Zonnepanelen PV2 Current: None>, <Entity Zonnepanelen Daily Production: None>, <Entity Zonnepanelen Total Production: None>, <Entity Zonnepanelen Total generation time: None>, <Entity Zonnepanelen Today generation time: None>, <Entity Zonnepanelen Output active power: None>, <Entity Zonnepanelen Output reactive power: None>, <Entity Zonnepanelen Grid frequency: None>, <Entity Zonnepanelen L1 Voltage: None>, <Entity Zonnepanelen L1 Current: None>, <Entity Zonnepanelen L2 Voltage: None>, <Entity Zonnepanelen L2 Current: None>, <Entity Zonnepanelen L3 Voltage: None>, <Entity Zonnepanelen L3 Current: None>, <Entity Zonnepanelen Inverter status: None>, <Entity Zonnepanelen Inverter module temperature: None>, <Entity Zonnepanelen Inverter inner temperature: None>, <Entity Zonnepanelen Inverter bus voltage: None>, <Entity Zonnepanelen PV1 voltage sample by slave CPU: None>, <Entity Zonnepanelen PV1 current sample by slave CPU: None>, <Entity Zonnepanelen Countdown time: None>, <Entity Zonnepanelen Inverter alert message: None>, <Entity Zonnepanelen Input mode: None>, <Entity Zonnepanelen Communication Board inner message: None>, <Entity Zonnepanelen Insulation of PV1+ to ground: None>, <Entity Zonnepanelen Insulation of PV2+ to ground: None>, <Entity Zonnepanelen Insulation of PV- to ground: None>, <Entity Zonnepanelen Country: None>, <Entity Zonnepanelen Fault 1: None>, <Entity Zonnepanelen Fault 2: None>, <Entity Zonnepanelen Fault 3: None>, <Entity Zonnepanelen Fault 4: None>, <Entity Zonnepanelen Fault 5: None>, <Entity Zonnepanelen status_lastUpdate: None>, <Entity Zonnepanelen status_connection: None>]
2023-02-11 11:44:33.092 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Starting to query for [2] ranges...
2023-02-11 11:44:33.099 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [0 - 39]...
2023-02-11 11:44:33.099 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a517001045000020dafe1e02000000000000000000000000000001030000002845d4c915
2023-02-11 11:44:33.417 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a563001015001f20dafe1e0201d342f502c4010000bd2ef2600103500002000000000000000000000a4e001e1457001d0008000f001200001387094c00600946005e095d005e00003f1c000038ae002400a00018002819730a3f1441003c00000001000004eb0bcb099d000749128415
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [0 - 39] succeeded
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [261 - 276]...
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a517001045000020dafe1e02000000000000000000000000000001030105001055fbee15
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a510001015002020dafe1e0201d442f502c5010000bd2ef26005008315
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] invalid modbus crc
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
[... cut ...]
2023-02-11 11:46:37.590 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry 192.168.1.73 for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 533, in async_unload
    result = await component.async_unload_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 28, in async_unload_entry
    hass.data[DOMAIN].pop(entry.entry_id)
KeyError: 'solarman'
`

This is the debug-log of the Solarman component. I filtered the other stuff out of the file, but hopefully I have not deleted to much.

lukasniemeier, 2023-02-11:
@JHurk thanks! Judging by your debug-log (see relevant excerpt) you seem to experience the same problem as the others, which means there is hope :+1: until a next version is released you may try to replace your local `sofar_lsw3.yaml` with the updated one already. Grab it from [here](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/sofar_lsw3.yaml).

```
2023-02-11 11:44:33.099 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [0 - 39]... 
2023-02-11 11:44:33.099 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a517001045000020dafe1e02000000000000000000000000000001030000002845d4c915 
2023-02-11 11:44:33.417 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a563001015001f20dafe1e0201d342f502c4010000bd2ef2600103500002000000000000000000000a4e001e1457001d0008000f001200001387094c00600946005e095d005e00003f1c000038ae002400a00018002819730a3f1441003c00000001000004eb0bcb099d000749128415 
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [0 - 39] succeeded 
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [261 - 276]... 
2023-02-11 11:44:33.418 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a517001045000020dafe1e02000000000000000000000000000001030105001055fbee15 
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a510001015002020dafe1e0201d442f502c5010000bd2ef26005008315 
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] invalid modbus crc 
2023-02-11 11:44:34.209 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] Querying [261 - 276] failed, invalid response packet.
```

JHurk, 2023-02-12:
@lukasniemeier Thanks, this did the trick.
After updating I edited the sofa_lsw3.yaml file and restarted HA.
After that everything was working again, even on 1.4.2
Looking forward to a 1.4.3 were this is permanently fixed.

heebo1974, 2023-02-12:
Yes, working here too with the new sofar_lsw3.yaml. :)

JHurk, 2023-02-17:
Fixed in 1.4.3, thanks!

closed 2023-02-17 by JHurk