StephanJoubert/home_assistant_solarman #192: All sensors unavailable for Sofar 6.6 KTL-X

mhoogenbosch, 2023-01-26:
Idk what happened but all my sensors are unavailable. I have removed the integration and reinstalled, but they stay unavailable, even after waiting for at least 10 minutes.

The system itself works just fine, the IP address is correct.

Maybe it is exactly the same as https://github.com/StephanJoubert/home_assistant_solarman/issues/189, but since the description is a bit different I created a new issue.

math322, 2023-01-30:
search in your inverter yaml definition for "validation" entry and remove/comment this entry

the unhandled exception is "ValueError"

mhoogenbosch, 2023-01-30:
> inverter yaml definition for "validation" entry and remove/comment this entry

I have set up the integration through the GUI. 

When I go to config/custom_components/solarman/inverter_definitions I can edit the sofar_lsw3.yaml, but when searching for validation it doesn't find anything. 

Am I looking into the wrong file?

dsgepers, 2023-01-30:
You can also try downgrade to 1.4.1 by redownloading it through HACS and see if that works

math322, 2023-01-30:
I use the deye_4mppt.yaml and the problem was gone after disabling the validation lines. Version 1.4.1 doesn't support my inverter.

mhoogenbosch, 2023-01-30:
For some reason it doens't seem to download 1.4.1. It keeps downloading 1.4.2. I'll try and see if I can get this going.

dsgepers, 2023-01-31:
You can choose the redownload option in HACS:

[image]

I had the issue with it redownloading 1.4.2. The solution for me was to be more patient after switching versions in the dropdown and wait for the dropdown to become available again after switching before clicking Download.

[image]

Uriziel01, 2023-02-02:
@mhoogenbosch Did you manage to solve your issue? I'm having the exact same one with `sofar_lsw3`

mhoogenbosch, 2023-02-03:
> @mhoogenbosch Did you manage to solve your issue? I'm having the exact same one with `sofar_lsw3`

No unfortunately not. I tried with the instructions above but still the unknown value. I know for sure I have 1.4.1 installed because it suggest to update to 1.4.2.

I found a different integration , which isn’t so nice as this one but at least that one works. This can be found here: https://github.com/MaGumagus/local_sofar_inverter

Hopefully the integration gets some love, more than willing to help , if I need to provide any logs please give some pointers to which.

HubertPodgorski, 2023-02-06:
Same issue here, i just don't know if it was working before as it is my initial setup.

Every entity is unavailable. 
Dd You try to do any workarounds with current integration. Don't want to install any other

mhoogenbosch, 2023-02-06:
> Same issue here, i just don't know if it was working before as it is my initial setup.
> 
> Every entity is unavailable. Dd You try to do any workarounds with current integration. Don't want to install any other

Yes, I tried installing 1.4.1 but doesn't work for me. Haven't tried earlier versions before 1.4.1, so maybe you could try those?

Uriziel01, 2023-02-06:
I've installed 1.4.1 and after a HA restart it's working again. Would love to help creators here by doing a research on the exact commit that broke the integration, unfortunately I do not have much spare time for the next few days.

HubertPodgorski, 2023-02-06:
How do i see if they are available or not? Status unknown on all of them, but never been known so cant confirm if its working. Got installed 1.4.1, not sure how long it takes entity to get the data

Uriziel01, 2023-02-06:
Few minutes usually (I think default is 5minutes), just remember the inverter has to be ON, so no testing at night or during solar eclipse 😄

HubertPodgorski, 2023-02-06:
Yeah, so mine does not work with 1.4.1 unless i should pick different config file. Which one are You guys using?

mhoogenbosch, 2023-02-06:
Mine doesn't work with 1.4.1 either. I don't know what happened, but it used to work fine.

HubertPodgorski, 2023-02-06:
With my Sofar HYD6KTL, with sofar config file i see "something" but production is 0. Also i see no entity for grid output/consumption and input, i see only solar entities

mhoogenbosch, 2023-02-16:
Updated to 1.4.3 no change.

Same issue: https://github.com/StephanJoubert/home_assistant_solarman/issues/215

sanderbaks, 2024-09-07:
@mhoogenbosch Hey Martijn! I assume you are Dutch (or Belgian). Did you manage to get it to work sofar?  I am having the same problems for quite some time now :). I really want to get it to work..

mhoogenbosch, 2024-09-07:
You are correct. From the Netherlands. I got things going but with https://github.com/davidrapan/ha-solarman instead of this integration. Works great.

sanderbaks, 2024-09-07:
I never saw that one!  I will try it now!

sanderbaks, 2024-09-07:
And can you tell me what config file you used?Mine is still not correctly working. The country is flipping between germany and denmark ( so read wrong ) and also all other values are all over the place. ----------------Sander BaksCommercial & editorial photographerwww.sanderbaks.comwww.hotelcreatives.comT: 06-46188558

mhoogenbosch, 2024-09-07:
[image] I've got this

sanderbaks, 2024-09-07:
> Upload.from.GitHub.for.iOS.png (view on web) I've got this

And what is your inverter? I got the 12kltx-3gWith the lsw3 wifi logger----------------Sander BaksCommercial & editorial photographerwww.sanderbaks.comwww.hotelcreatives.comT: 06-46188558

mhoogenbosch, 2024-09-08:
One advise first to remove your mobile phone number(!)

The one I got is mentioned in the first post. But don't have WiFi. Got it Ethernet connected.

sanderbaks, 2024-09-08:
Could you please say again what inverter you have? I have a 12ktlx-g3
So 3 phase. 

I do have a connection now. And i do have values, but they dont make sense. 

So i guess i need some new translation of the variables.

mhoogenbosch, 2026-10-11:
Closing on my side: the Sofar inverter this report was about is being replaced, so I can no longer test. Thanks to everyone who shared workarounds in this thread — reopen if it still helps to track it.

closed 2026-10-11 by mhoogenbosch