StephanJoubert/home_assistant_solarman #193: Afore inverter with Solarman integration

Arnesen, 2023-01-30:
This integration looks great! And I am looking to integrate my Afore Solar inverter. https://www.aforenergy.com/product/show/id/5
I am currently using the Solarman app on my phone to get data successfully, and I have installed and gotten access to my wifi-inverter thing. I have also integrated it through HACS and you plugin. But from my understanding, you do not have a inverter definition for my inverter yet. But I want to contribute, and I want to try to write the definition if you can point me in the right direction. 

How would I go about working out parameters that is defined in the .yaml? And is there a good workflow for iteration on that locally without installling and uninstalling the HACS addon? 

secondly, my wifi-thing has a higher serial. 35xx. Could that cause issues? 
[image]

Any help is appriciated, but feel free to close this issue there is no options for me :)

RazorCopter, 2023-08-11:
news?

RazorCopter, 2023-08-11:
I contacted the manufacturer who provided me with all the technical documentation that I am attaching, if someone manages to integrate it it would be great!

https://drive.google.com/drive/folders/1b9ATSZeUsNePeMUKsWuGoqX8j_7MJNeg?usp=sharing

ketoha, 2024-05-12:
is there any progress? Struggling with afore and 27x serial