StephanJoubert/home_assistant_solarman #195: Querying registers fails for Deye SUN600G3-EU-230

Byter09, 2023-01-31:
Hello,

I own a Deye SUN600G3-EU-230, that is currently connected to a single panel with 420W for testing purposes.

Here's my current configuration:
```
HA version: Home Assistant 2023.1.7
Inverter firmware: MW3_16U_5406_1.53
Device S/N: 416*******
Port: 8899
Slave ID: 1
Parameters: deye_4mppt.yaml
```

I am unsure if I'm setting things up wrong or whether there's a problem with the inverter, but no matter what parameters I try, I always get the following in my HA logs:
```
2023-01-31 13:02:33.076 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.deye_pv1_voltage is taking over 10 seconds
2023-01-31 13:01:25.062 WARNING (SyncWorker_7) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.
2023-01-31 13:00:25.332 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.
2023-01-31 13:02:39.275 WARNING (SyncWorker_9) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.
2023-01-31 13:03:30.467 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.
```

At this point I've run out of ideas and even thought about getting API access to the solarmanpv site, however I'd prefer to be able to collect data without relying on a cloud. Once I set up local data collection I'll even turn off the cloud access with my firewall. Not a big fan of devices phoning home.

This might be related to  #130 but I'm not sure, so I'm opening a new issue. I'm sorry if that's wrong.

If there's any tests you need me to perform, I would be happy to run those.

Thanks for taking the time to look at this!

hparusel, 2023-02-01:
Hey there,
I got my Deye SUN600G3-EU-230 (sames serial no.) two days ago. Unfortunately, I experienced the same issue like you. I was hoping that a downgrade to 1.4.1 might solve the issue but it did not...

math322, 2023-02-01:
you can try this https://github.com/StephanJoubert/home_assistant_solarman/issues/192#issuecomment-1408446447

Byter09, 2023-02-01:
Hahaha! This actually worked. Could it be, that this validation prevents freshly set-up installations from being considered "valid"?

hparusel, 2023-02-01:
Hmmm did not work for me.... Which Inverter-Yaml Do you use? Have u completely commented out all there lines? Super happy about your help because I do not want to use the China cloud... 
[image]

Byter09, 2023-02-01:
Just for the convenience of future readers:
1. take `deye_4mppt.yaml` and copy it to `custom_parameters.yaml`
2. search for `validation` and comment out the whole part to look like this:
```
#      validation:
#        min: 0.1
#        invalidate_all:
```
3. Set up the integration as described in the readme, but now choose `custom_parameters.yaml`

This made it work for me. :)

hparusel, 2023-02-01:
Thanks for your detailed guide.... For me it is still not working. Nevertheless, the error message chagend to: 

> 2023-02-01 12:37:10.925 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.

lukasniemeier, 2023-02-01:
@hparusel  looking at the frames that are coming from your inverter, this looks similar to https://github.com/StephanJoubert/home_assistant_solarman/issues/187#issuecomment-1410900720

CybDis, 2023-02-02:
Same problem here, same fix worked (removing the validation lines)

hparusel, 2023-02-02:
@CybDis May I ask which firmware is installed on your inverter?

Byter09, 2023-02-02:
That's the one that was installed when it arrived. I upgraded manually using https://github.com/dasrecht/deye-firmware/blob/main/0_D0002_18/MW3_16U_5406_1.53.bin

distancerunner, 2023-03-13:
I use firmware 1.53 on the deye and the 2Mppt Config. It works well and without any problems since a few days.

DHLF, 2023-03-19:
**Deye SUN600G3-EU-230 no(empty) data issue**

Maybe totally wrong though about this issue..
I played around with some stuff in the hidden config(config_hide.html) of the microinverter to deactivate the access point:
If the Working mode is set to TRANSPARENCY(instead of DATA Collection) there will be only EMPTY Values either if you access the normal status page or using console command:
curl -u username:passwort ip-address/status.html | grep "$dat ="
So IF no data is shown(while inverter is connected) have a look at the *secret* page.

Perhaps wrong config file used(for deye600 the deye_2mppt.yaml file should be used as other definitions may contain additional sensors  which are NOT present in your device)

migo007, 2023-04-16:
@hparusel 
I saw the same "invalid modbus crc" error as you in my logs, and also couldn't fix it by removing the validation line. 
In my case, it turned out that my serial number shown in the web interface was wrong. I retrieved the serial number using the autodiscovery from:
[Script from pysolarmanv5](https://github.com/jmccrohan/pysolarmanv5/blob/main/utils/solarman_scan.py)
and it worked immediately.

k2iuh, 2023-07-02:
I had the same issue and found what was wrong for me. I have  aSUN-M80G3-EU-Q0 and was using the wrong serial number. Do NOT use the inverter serial number, but the device serial number, to query your inverter. I looked at the modbus messages directly and this is what the error code reported (wrong serial). Unfortunately, this library does not handle this error code.

eustaquiobp, 2024-11-01:
Hi guys, I had the same problem here. In my case I just change the yml for deye_4mppt.yaml and fix the issue.
Tks for instructions.