StephanJoubert/home_assistant_solarman #201: HA Solarman  disconnect continuously all the  day

lula73, 2023-02-03:
Hi  
anyone  have the same problem? As you can see in the image,  the solarman connect and disconnect continously all the day: 

[image]

I've tried : 

-  check connection problem : wi-fi is always  around 70%    and my Raspberry see continously  the inverter

-   I turn off the sending data to china servers  using   /config_hide.html    panels
   
-  rebooting the stick 
   
-   lookin in the HA Logs. Nothing interesting, sometime  i see  :  _WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_produzione_energia is taking over 10 seconds_

so, the issues remain.    I think it's a  problem of  wi-fi  stick  software becouse sometimes  pinging the Inverter IP  respond,   but the control panel  do not respond. So seem software problem in the stick.

Maybe it's better to set  scan_interval on  10 minutes instead  1 minute ?

Thnks in advance

Simoverdex, 2023-02-18:
I have the same problem with ZCS YHD 6000-ZSS and logger with LSW3_15_FFFF_1.0.57 firmware.
The wifi quality of my logger is 90%, so i don't think is a wifi problem. Maybe the logger is not be able to work with many request every minute.

Spangeac, 2023-02-20:
I have the same issue. As mentioned above, when I log into the Logger IP address it shows "Signal Quality" is 100%.
[image]

And the History shows often disconnection.

[image]

Any suggestions will be very welcome.

lula73, 2023-02-20:
Hi guys,  anyone has tested the new version 1.4.3 ?  
I think it's someting related to the socket connection that remain open and create problem to the wi stick. I will try in the next days to upgrade

Spangeac, 2023-02-20:
Hi, I have been testing here and had the HA "History" log running on Solarman_status_connected" and the HA History log updated every minute. When it indicated "disconnected" I could see its AP on the WiFi list but could not log into its IP address. 

Can you please let me know what Version 1.4.3 refers to 

[image]

lula73, 2023-02-22:
_"I could see its AP on the WiFi list but could not log into its IP address"_

IT's the same for me.  If you try to restart the stick Logger and/or  HA  you can see that come back to "connect". So i think the problem is that HA  send too much request to  Stick  logger  and generate some "memory" problem, and maybe the stick after some period  restart itself or shut down all the open connection. 

the  Version 1.4.3  i mean of home assistant solarman    https://github.com/StephanJoubert/home_assistant_solarman/releases
 I'm actually running 1.4.1   (i installed in december). Maybe in the next days i will try the update (need to backup all)

Spangeac, 2023-02-22:
Hi,
I saw a notification on my Android Solarman App that an update was taking place 21 Feb 2023 at around 6pm so I dont know if this is co-incidence. But I had a look at Wifi last night and I have a Range Extender. I saw that the Range Extender was not close enough to the router so I located it half way between Router and Logger. % on the logger went down from 100% to 74% but the Extender now has a good input signal. I also suspect that the signal quality indicated is the signal strength and not the quality / integrity of the data stream. 
Since 6AM there has been no break in the "Connection status" 
Maybe coincidence. 
[image]

PS. I have been on 1.4.3 for a few days and it was still loosing connection. Maybe good to do the Solarman update anyway.