StephanJoubert/home_assistant_solarman #202: Energy panel problem with consumption

asiviero76, 2023-02-04:
[image]
On my enery panel, is like the consumption from 00:00 till when there is the last production will be accumulate ... how to solve this problem? 
I'm using sofar_g3hyd.yaml configuration for a Zucchetti Hybrid inverter... all the rest are working fine.
Help are really appreciated
thanks
andrea

udivankin, 2023-02-04:
I second this, the value for active power should be zero instead of unknown at night (and after reboot) to get energy dashboard to work. I believe introducing optional default_value field in inverter definition's register would help.

asiviero76, 2023-02-04:
> I second this, the value for active power should be zero instead of unknown at night (and after reboot) to get energy dashboard to work. I believe introducing optional default_value field in inverter definition's register would help.

Thanks for suggestion, but I really don't know how to do this

asiviero76, 2023-02-07:
> To prevent this, daily sensor must be set to "0" everyday at 00:00. I tried to fix this with changing "total" to "total_increasing" in yaml file (from configuration of entity, at bottom there is name of yaml file located in custom component folder) under Daily production. I will update tomorrow if this helped.

Will be very helpful! Thanks

oziee, 2023-02-21:
@Nubstex any update?