StephanJoubert/home_assistant_solarman #204: update integration on Home Assistant remove custom_configuration.yaml

linuxdep, 2023-02-06:
Hi,
after update of integration from solarman my custom_configuration.yaml was removed. 
I thought a custom configuration should survive every update, neither be deleted nor overwritten. Can you please take a look, thank you.

rolandsteinmeyer, 2023-02-19:
also happened again upgrading to V1.4.3 with HACS.

joekky, 2023-02-23:
@linuxdep and @rolandsteinmeyer do you mind discribing in few words how to load a custom_config.yaml file? I'm new to HA as a whole. I want to create a shorter interval. Guess I did it write

rolandsteinmeyer, 2023-02-24:
custom_config.yaml is explained here: [https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md](url) But you can not change the request interval here. AFAIK the logger gets data from inverter every 1-2 minutes, so no sense to do more requests to the logger. If ypu need real-time ddata you need to add a seperate energy meter (e.g. shelly PM).

linuxdep, 2023-02-26:
Or use a other RS485/Modbus communication applikation.