StephanJoubert/home_assistant_solarman #213: sofar_hyd3k-6k EP with lsw3 wifi logger - Querying registers [512 - 597] failed, aborting

richardob50, 2023-02-11:
Inverter was installed during the week and have been trying to get this integration to work since.

I have tried both the LSW3 and sofar_hyd3k-6k lookup (logger and inverter) but i am seeing the below error in the log for both. I am only a novice user of HA so I'm not sure how to get more detailed info to share

This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:175
Integration: Solarman (documentation, issues)
First occurred: 15:23:35 (6 occurrences)
Last logged: 15:26:05

Querying registers [512 - 597] failed, aborting.

sjmtlewy, 2023-03-07:
I have a new HYD 3600-ES with the LSW3 logger. I'm getting similar failures:

2023-03-07 21:52:38.941 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_pv1_power is taking over 10 seconds
2023-03-07 21:52:41.062 WARNING (SyncWorker_2) [custom_components.solarman.solarman] Querying registers [3 - 112] failed, aborting.

cvandesande, 2023-07-30:
Using solar_g3hyd resolved this for me with my Sofar HYD 6000-EP