StephanJoubert/home_assistant_solarman #217: this will fail in version 2023.3.

Macher00763, 2023-02-13:
Log Message:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:77
First occurred: 18:40:44 (1 occurrences)
Last logged: 18:40:44
Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for solarman using this method at custom_components/solarman/__init__.py, line 18: hass.config_entries.async_setup_platforms(entry, PLATFORMS)

fabiobrolo, 2023-02-13:
Indeed is failing:

Logger: homeassistant.helpers.entity
Source: custom_components/solarman/solarman.py:234
Integration: Solarman ([documentation](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md), [issues](https://github.com/StephanJoubert/home_assistant_solarman/issues))
First occurred: 19:49:17 (40 occurrences)
Last logged: 19:51:45

Update for sensor.gerador_input_active_power fails
Update for sensor.gerador_output_apparent_power fails
Update for sensor.gerador_output_active_power fails
Update for sensor.gerador_output_reactive_power fails
Update for sensor.gerador_inverter_id fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 548, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_device_update
    raise exc
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solarman/sensor.py", line 140, in update
    self.inverter.update()
  File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 190, in wrapper
    result = method(*args, **kwargs)
  File "/config/custom_components/solarman/solarman.py", line 175, in update
    self.get_statistics()
  File "/config/custom_components/solarman/solarman.py", line 234, in get_statistics
    sock.close()
UnboundLocalError: local variable 'sock' referenced before assignment

5h4mr0Ck, 2023-02-23:
Is there any workaround for this? Great integration but the last HA update has broken it. Do I need to roll back or is there something I can change locally?

joschmid, 2023-03-13:
Is there an update for this? Basically you should update to `2023.03.1 or later` because there was a [security issue](https://www.home-assistant.io/blog/2023/03/08/supervisor-security-disclosure/), so a solution for the problem with the newer version would be important.

Capone1983, 2023-03-23:
Is there a plan to fix this issue?

fabiobrolo, 2023-03-23:
Try redownloading the main version instead of the latest release v1.4.3.

cseyda, 2023-04-01:
the initial issue should be fixed with this commit https://github.com/StephanJoubert/home_assistant_solarman/commit/0cd2e8d0f98133682522a7e622e29373b5410227 in release 1.4.3