StephanJoubert/home_assistant_solarman #219: Deye-Sun-600 Wrong false measurement

Teyro, 2023-02-19:
Good day,

I am writing with the help of DeepL because my English is not so good. I have a Deye solar system with the model number SUN-600 and I have connected my inverter to Home Assistant with the following settings.
[image]

Data also appears to be evaluated and displayed. Unfortunately, my statistics under "Energy" reset themselves at zero o'clock, so that minus values arise. In addition, I see no progression in the bar chart, but always only "the last value".  I wish so much that I can see a progression, as with the consumers. How can I set this up?
[image]

Kind regards and thank you very much!

ohoeltke, 2023-02-22:
You only have to configure one source for the Energy Dashboard, and it has to be one that is always incremented and not reset on a daily basis. (see: https://github.com/StephanJoubert/home_assistant_solarman/blob/main/energy.md)

If you remove your `Balkonkraftwerk Daily Production` and only use `Total Production` everything will be fine ;)

distancerunner, 2023-03-03:
I#m helping in german:
Ich werde in den kommenden Wochen auch den SUN600 einbinden.

Wie auch beim Stromzähler kannst du besser die Totale Produktion einbinden. Also ein Wert, welcher täglich größer wird. Wie auch beim Stromzähler, wo der Wert auch täglich größer wird, werden die Daten dann passend angezeigt. Wenn es bei dir geklappt hast, kannst du dein Ticket hier schließen.

distancerunner, 2023-03-12:
I was able to add the total production value to the energy dashboard. The deye inverter api is working fine.

ede-kowalski, 2023-06-24:
Hi all,
what happens at your dashboard in the morning?
My dashboard shows then the initial value bcs its gone to zero in the night.
Any hint to solve that?

Thanks,
Ede