StephanJoubert/home_assistant_solarman #221: Sofar HYD5000-ES Battery Type and SoH

slaughtered-lamb, 2023-02-20:
Hi

For your fantastic integration to work correctly with my Sofar HYD5000-ES I must make the following changes to the sofar_hyd3k-6k-es.yaml

    - name: "Battery SoH"
      class: "battery"
      state_class: "measurement"
      uom: "%"
      scale: 1
      rule: 2
      registers: [0x0237]
      icon: 'mdi:battery'

    - name: "Battery Type"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [0x10B0]
      icon: 'mdi:battery'
      lookup: 
      -  key: 0x0000
         value: "AMASS"
      -  key: 0x0001
         value: "LG SMART"
      -  key: 0x0002
         value: "PYLON"
      -  key: 0x0003
         value: "CATLL"
      -  key: 0x0004
         value: "Alpha. ESS"
      -  key: 0x0005
         value: "General Lithium"
      -  key: 0x0006
         value: "DEFAULT"

Hope this helps
Thanks
Karl

warner83, 2023-03-26:
What error did you have?

slaughtered-lamb, 2023-05-20:
There was no error, just incorrect information

Nusserdt, 2023-08-11:
I have a **HYD 5000-EP / 5kW** and the solar integration with `sofar_hyd3k-6k-es.yaml` profile returns me only unknown values:

[image]

I also wondering about the local web interface of the wlan stick. The values are totally wrong:

[image]

Firmware version: **LSW3_15_270A_1.53**