StephanJoubert/home_assistant_solarman #222: Like to contribute working inverter definition for SUN600G3-US-220/EU-230

mostrowi, 2023-02-22:
First of all thanks for providing the Solarman Integration, great work! 

I believe there is a significant interest to use your solarman integration at least in Germany from folks using these smaller two module Microinverters. Since I was in the same situation recently I tried to get the existing inverter_definitions to work, but without success. So I hacked together a new deye_2mppt.yaml definition based on deye_4mppt.yaml and changed const.py to pick up the new definition. I have this definition working fine since a few month now. I would like to contribute that definition to the community if wanted. I had to rename the file to be able to upload it...
[deye_2mppt.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/10801525/deye_2mppt.yaml.txt)
This would also fix the issue to re-apply the changes manually for each update of solarman. Let me know if you need any additional information about the used Inverter model or anything else.

ohoeltke, 2023-02-22:
i just wanted to create a PR for the same thing.
I also have to remove the validation part for the SUN600G3 #195 

will you raise a PR or should I do it?

mostrowi, 2023-02-22:
Go ahead and raise the PR

ohoeltke, 2023-02-22:
i left the validation in place, since i don't have problems anymore. Maybe the inverter didn't had a total production > 0.1 when i initially added it to HA.

PR #224 (2023-02-23): added configuration for Deye SUN600G3
I modified the configuration for SUN2000G3 to match the SUN600G3. 
It will probably also work for the 800G3, but i don't have a device to test. 
will resolve #222
closed 2023-02-23 by StephanJoubert