StephanJoubert/home_assistant_solarman #226: Sofar inverter module LSW3_15_270A_1.38 reset

jeffersonbernardo1981, 2023-02-23:
Expensive,

I have a problem with the integration of the solarman repository, I followed the step by step according to the documentation, my Inverter is Sofar, the module version is LSW3_15_270A_1.38.
Some entities inform me of some and tell me that they are connected according to the photos below, but most entities are all reset, could it be that someone has already gone through this and managed to solve it and can give me any tips?
[image]
[image]
[image]

gushcs, 2023-04-18:
try this code 
https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650

piratini, 2023-08-17:
> try this code https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650

I was having the same problem using a Sofar 4KTLM-G3 inverter. Using the code cited ([6ktlm-g3.yaml](https://gist.github.com/gushcs/223ca2c61af2d345238d286746617650)) I was able to access the inverter data.

Thank you.