StephanJoubert/home_assistant_solarman #229: Disable logging by night/sunset whatelse

wunderbaum, 2023-02-24:
It works great, but my inverter is only sending data when the solarpanels produce some load. So, when the sun sets, there is no data any more.
Now solarman is foaming the logs that it cannot reach the device any more.
Would be nice to have a switch to disable logging completly as its obvious that there is no data at least after sunset until sunrise.
I mean, after setting up everything, its not necessary to log any more.

distancerunner, 2023-03-13:
Hi. I have some logs for this abnormal situation. I think this abnormal values are also a problem for the energy dashboard. See the screenshot attached: It stopps after 23:59 and doesnt show values on the next day. If I restart the docker container, all the values from 0:00 are shown correctly:

`2023-03-12 06:16:38.970 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying inverter 4179952036 at 192.168.178.96:8899 failed on connection start with exception [OSError]

2023-03-12 06:17:39.051 WARNING (SyncWorker_0) [custom_components.solarman.solarman] Querying inverter 4179952036 at 192.168.178.96:8899 failed on connection start with exception [OSError]

2023-03-12 06:18:38.054 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.

2023-03-12 06:19:38.037 WARNING (SyncWorker_5) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.

2023-03-12 06:20:38.103 WARNING (SyncWorker_4) [custom_components.solarman.solarman] Querying registers [3 - 128] failed, aborting.`

[image]

distancerunner, 2023-03-14:
I can confirm it: My template in issue #116 was able to run the energy dashboard over midnight without problems. This morning, my data was continuously shown after the day change at midnight.