StephanJoubert/home_assistant_solarman #232: Deye 12K-SG04LP3 Total PV Power is missing

hasant73, 2023-02-28:
Many thanks for sharing your project. It works very well for me. 
I only noticed that I cannot get the "Total Solar Power" from the logger. 
PV1 Power and PV2 Power are available and can be of course added together. But the logger also provides the sum of both. 

Thanks.

ondrejbartas, 2023-03-02:
Hello i have created custom value in ha helpers which combines both values

hasant73, 2023-03-03:
I did the same but somehow it is showing 0 all the time.
Can you share your config?

`sensor:
  - platform: template
    sensors:
        solar_power_total:
            friendly_name: "Solar Power Total"
            unit_of_measurement: 'kWh'
            value_template: '{{ states("entity:sensor.solarman_pv1_power") | float(default=0) + states("entity:sensor.solarman_pv2_power") | float(default=0) }}'`

Bodypull, 2023-03-10:
You can create a "Min / Max" helper where you set "Statistic characteristic" to "Sum". There you can add PV1 and PV2 as sensors to build the sum.

PV-Joe, 2023-03-10:
[image]

hasant73, 2023-03-10:
My helpers look different

[image]

Bodypull, 2023-03-10:
Min / Max... is the one you want

Bodypull, 2023-03-10:
Select "Sum" at the very bottom:

[image]

hasant73, 2023-03-10:
Yes understood. But in my selection there is no "Sum".

[image]

hasant73, 2023-03-10:
What version of HomeAssistant do you have?

Bodypull, 2023-03-10:
It works on 2023.1.2 and 2023.3.1

Can you try switching to english for testing?

PV-Joe, 2023-03-10:
> My helpers look different
> 
> [image]

> My helpers look different
> 
> [image]

Update Home Assistant to the latest version and you will find the “Combine” Helper.