StephanJoubert/home_assistant_solarman #237: Serial Number 390

K3vb3rt, 2023-03-05:
Hey guys,

i have freshly installed my Deye SUN 800 and try to connect it to my home assistant instance. 
Periodically it looses the connection. In the tutorial I read:

"Check the version of the solarman logger. If the serial number starts with 17xxxxxxx or 21xxxxxxx (protocol V5), the component should work. If not, you may need to try the component for V4 of the protocol mentioned above." 

**My serial starts with 390**. What can I do? Which lookup_file: should I use? 🙂

Thank you!

gejomo, 2023-03-12:
Hey guy’s,
I have the same problem. The serial number of my Deye logger starts with 417….

prankousky, 2023-07-05:
Same here. Mine starts with `3914`.

closed 2023-12-29 by K3vb3rt