StephanJoubert/home_assistant_solarman #238: Add ability to change scan_interval from UI

vakius, 2023-03-06:
As far as I understand, to change the scan_interval from default 60 seconds to another value I have to create a sensor via configuration.yaml, because scan_interval  is not shown in config_flow.

Could you, please, add scan_interval to config_flow, or is there a reason not to do so?

PV-Joe, 2023-03-06:
I think the scan interval is set by the logger. As far as I know it can’t be set to a lower value by Home Assistant.

if you want real time statistics you have to pull the data with a cable.

onlyv8, 2023-03-07:
what's the data transfert protocol of LSW 3 logger ?

thanks for the answer

onlyv8, 2023-03-07:
PV Joe, you write that we can pull the data with a cable . is it possible ?

because, the data logger, i don't need it in this case: my router is 1m far my inverter !

and no lag after this. no ?

what is the LSW3 with db9 connector: rs323 or rs485 ?

juste need to buy an adapter with the right transfert protocol and ethernet connector , no ?

thanks you by advance for the answer

Eric

PV-Joe, 2023-03-07:
Look here:

https://solar-assistant.io/explore/deye#hero

It is a solution with a RS485 cable hooked to a Raspberry Pi fetching data in real time.

Hope it helps.

onlyv8, 2023-03-07:
PV Joe, 

Waouh !!! Thanks you so much.

I'll stude it !

Eric

onlyv8, 2023-03-07:
But this software do not support Sofar Solar inverter... Sniff !!

vakius, 2023-03-07:
I just asked to make this 'scan_interval' parameter available not only to manual configuration in yaml files, but to be able to set it in GUI too.

onlyv8, 2023-03-07:
But in my case, reduce scan_interval do not affect the data transfert...

PV-Joe, 2023-03-08:
In addition to that, the manual configuration guide of this integration states that there is a parameter `scan_interval | Time in seconds between refresh intervals` https://github.com/StephanJoubert/home_assistant_solarman#parameters

Great news and thx for pointing out!

Now I’m curious how to set the “scan_interval” parameter to 5 seconds.

I’ve used the configuration by GUI and got no idea where and how to insert the parameter for my Deye Inverter without switching completely to manual configuration.

Could someone give me hint?

zbyndatvrd, 2023-03-22:
> 

I've tried the manual configuration, set scan_interval to 5. 
Now all values are refreshed every ~30s. 

Are you sure, the 5 seconds interval is possible with @StephanJoubert's solution? 

Thanks!

vakius, 2023-03-22:
> Are you sure, the 5 seconds interval is possible with @StephanJoubert's solution?

I've used [jlopez77's](https://github.com/jlopez77/DeyeInverter) solution as a standalone script that connects to the Logger, reads info and sends it to MQTT broker and exits. The whole process takes less than a second to run. For testing I've set a pause of 1 second between runs and the data was retrieved reliably with such a small delay.

Unfortunately І don't know about inner workings of the @StephanJoubert's solution.
I just wanted to make the point that a 30-second interval is not a limitation on the inverter's or logger's part.

MarcFu, 2023-03-27:
Hello,

I am also using a Deye SUNG3 micro inverter and would appreciate configuring the _scan_interval_ via the config flow. I also went the way via UI. The energy dashboard in HomeAssistant is very fragile when underlying entities are modified. So being able to change the _scan_interval_ from the UI would help a lot.

Greetings,
Marc

rkistner, 2023-03-29:
The interval is currently controlled by two things:
1. The Home Assistant default polling interval, that controls when each individual entity is updated.
    * Not configurable when you configure the sensor via the UI.
    * Can manually trigger a refresh by using the `homeassistant.update_entity` service on a schedule, but you have to do this for every entity.
2. The hardcoded `MIN_TIME_BETWEEN_UPDATES` that controls the minimum time between requesting updates from the inverter.

Currently there isn't an easy way to adjust either of those, but I've adjusted the scan interval to 15 seconds with the following changes to the code:

```diff
diff --git a/custom_components/solarman/const.py b/custom_components/solarman/const.py
index 81da34c..b05c014 100644
--- a/custom_components/solarman/const.py
+++ b/custom_components/solarman/const.py
@@ -21,7 +21,7 @@ LOOKUP_FILES = [
     'custom_parameters.yaml'
 ]
 
-MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=30)
+# Must be less than SCAN_INTERVAL
+MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=10)
 
 CONF_INVERTER_HOST = 'inverter_host'
 CONF_INVERTER_PORT = 'inverter_port'
diff --git a/custom_components/solarman/sensor.py b/custom_components/solarman/sensor.py
index a5ad8ab..ace417e 100644
--- a/custom_components/solarman/sensor.py
+++ b/custom_components/solarman/sensor.py
@@ -23,6 +23,7 @@ from .services import *
 _LOGGER = logging.getLogger(__name__)
 _inverter_scanner = InverterScanner()
 
+SCAN_INTERVAL = timedelta(seconds=15)
 
 def _do_setup_platform(hass: HomeAssistant, config, async_add_entities : AddEntitiesCallback):
     _LOGGER.debug(f'sensor.py:async_setup_platform: {config}') 
```

The inverter and WiFi adapters are likely capable of higher update rates (it's just two modbus requests to read a bunch of registers), but I haven't tested how far you can push it.

PV-Joe, 2023-03-30:
Hi Ralf,

your solution sounds promising. 😄
Would you mind to explain how and where exactly I have to implement your code to get the 15 second polling interval?
Some screenshots could help to understand.

Thy in advance!

rkistner, 2023-03-30:
@PV-Joe The diff above shows the changes - new lines in green, removed lines in red, file names also highlighted.

Make sure you have direct SSH or other access to edit the files - if you make a mistake in editing, Home Assistant could fail to start up.

And note that you'll lose these changes when updating the repo via HACS.

onlyv8, 2023-03-30:
rkistner,

in wich file must we change these values ?

Thks

PV-Joe, 2023-04-01:
@StephanJoubert

Are you planning to integrate the solution by @rkistner to the UI? This would be a great feature and extend the polling closer to “real time”. ☀️

TheRealGreatOldOne, 2023-04-04:
I’d like to see the update solution by @rkistner (ralf) included. Whilst the 1 minute updates are much better than the SOLIS web based data that their app uses, you can still get the snapshot showing momentary data that happened just as the scan was done - leading to wifely confusion and endless questions about why it’s selling to the grid or doing something else unexpectedly… 🤪

gionaf, 2023-07-30:
+1
Please add the option to the UI so we won’t lose the changes to the code on update

gardiol, 2024-10-07:
We need this. My inverter bails out and the integration will print lots of errors in the logs (and missing points in the history data...) if this is anything lower than 30 secs...

For the time being i have manually modified the code, but we really need a better approach.

PV-Joe, 2024-10-07:
Try this more advanced and well maintained ha-Solarman integration by David Rapan.

This allows a scan intervall from down to 5 seconds and works 100% stable.

https://github.com/davidrapan/ha-solarman

perettigiuliano, 2024-12-21:
> Try this more advanced and well maintained ha-Solarman integration by David Rapan.
> 
> This allows a scan intervall down to 5 seconds and works 100% stable.
> 
> https://github.com/davidrapan/ha-solarman

Sorry @PV-Joe , I want to try that integration but after downloading it I cannot find it anywhere. COuld you help me