StephanJoubert/home_assistant_solarman #242: Inverter Alert Message LOOKUP

coserotondo, 2023-03-09:
I'm using the following lookup values for Inverter Alert but sometimes the outcome is literally the 'LOOKUP ' string instead of one of the value for the relative key. Does it mean that the received data is not in the keys provided?
`
    - name: "Inverter Alert Message"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [0x022B]
      isstr: true
      lookup: 
      -  key: 0
         value: "No error"
      -  key: 1
         value: "ID81 Internal temperature is too high"
      -  key: 2
         value: "ID82 AC frequency is too high"
      -  key: 4
         value: "ID83 Long distance load shedding"
      -  key: 8
         value: "ID84 Switch OFF ME3000SP remotely"
      -  key: 16
         value: "Reserved"
      -  key: 32
         value: "Reserved"
      -  key: 64
         value: "ID87"
      -  key: 128
         value: "ID85 Battery voltage below SOC"
      -  key: 256
         value: "ID86 Battery voltage is too low and cause ME3000SP to switch OFF"
      -  key: 512
         value: "Reserved"
      -  key: 1024
         value: "Reserved"
      -  key: 2048
         value: "Reserved"
      -  key: 4096
         value: "Reserved"
      -  key: 8192
         value: "Reserved"
      -  key: 16384
         value: "Reserved"
      -  key: 32768
         value: "Reserved"
      icon: 'mdi:wrench'
`

qfinucane, 2023-04-29:
Hi, Yes, its receiving a value that the LOOKUP cannot interpret / convert. I experienced the same with different registers. See closed issue #255.
I removed the lookup portion of the code and recorded the values. Then added the missing values to the Lookup. You will just need to research what the value means.

coserotondo, 2023-04-30:
Thanks, trying this approach to decode the message.

coserotondo, 2023-05-08:
Decoded the messages, no more LOOKUP.

closed 2023-05-08 by coserotondo