StephanJoubert/home_assistant_solarman #245: Deye 12K-SG04LP3: Querying registers [586 - 591] failed, aborting.

archi, 2023-03-11:
Installed via HACS and configured the integration to use the inverter IP (dhcp static assignment), logger serial number and the deye_sg04lp3.yaml.

Inverter serial# starts with 221...; logger with 272...
Firmware for both was updated by Deye a week ago.

Since there is nothing connected to the unit, most values should be 0.

However, the integration doesn't seem to work (all entities have an unknown value), and after restarting HA I get the following error in the log:

```
Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:173
Integration: Solarman (documentation, issues)
First occurred: 17:54:48 (1 occurrences)
Last logged: 17:54:48

Querying registers [586 - 591] failed, aborting.
```

When using the default yaml file, values can be read. But since the registers don't match, the values are wrong.

FatalSunrise, 2023-03-15:
I have same problem, but with different inverter. My inverter is SofarSolar 12KTLX-G3. The erro in homeassistant is: Querying registers [8195 - 8197] failed, aborting. I'm using another python code on my linux machine and I can read data from my inverter with it, so I don't think the logger or inverter is the problem. Also this integration used to work at least last summer.

anonid64, 2023-03-26:
Same issue with you
[image]

warner83, 2023-03-26:
Same here, model 1PH HYD5000

Metalage89, 2023-06-13:
Same problem since today with Deye 12k Hybrid Inverter:

2023-06-13 21:11:14.197 WARNING (MainThread) [homeassistant.components.sensor] Updating solarman sensor took longer than the scheduled update interval 0:00:30
2023-06-13 21:11:14.683 WARNING (SyncWorker_3) [custom_components.solarman.solarman] Querying registers [514 - 558] failed, aborting.