StephanJoubert/home_assistant_solarman #247: Switched from wireless lan logger LSW-3 to cable logger LSE-3

hasant73, 2023-03-20:
I first tried the LSW-3 logger and I could connect with home assistant to the logger and get data. 
But I don't have good WLAN in the cellar where the inverter is installed. So I switched from the wireless version to the LAN cable version called LSE-3. LSE-3 is reporting to solarmanpv website successfully. But Home Assistant doesn't want to connect anymore. Does somebody know if there is a setting I need to change or have any other idea what's wrong. 

I have changed the serial number of the logger in the config. Otherwise I use the deye_sg04lp4.yaml file. 
I entered the IP address of the logger and setup the network that it is always the same IP address. 
I can also reach the configuration web page of the logger.

MartinNeurol, 2023-03-20:
Hi, I second this -  got LSE3 wired logger and so far was unable to connect it to HASS at all. I Will gladly serve as a sandbox for LSE3 integration testing :) (SN 21xxxxxxxx...) firmware MW4C_14_CSIP_1.14

hasant73, 2023-03-20:
> 

It looks like the stick has port 8899 open:

`
Starting Nmap 7.80 ( https://nmap.org ) at 2023-03-20 15:43 CET
Nmap scan report for solarman-logger.fritz.box (192.168.178.125)
Host is up (0.0029s latency).
Not shown: 998 closed ports
PORT     STATE SERVICE
80/tcp   open  http
8899/tcp open  ospf-lite
`

Matz88, 2023-03-29:
I read in other issues that this integration supports only the WiFi dongles, some had success with the beta version, but most not. Search for "ethernet" in the issues. If you find a way, please tell me!

sanderbaks, 2023-04-03:
Ik have the same issue :(

sanderbaks, 2023-05-07:
I still did not get it to work? Did anyone find a solution?

Matz88, 2023-05-07:
If you have the LAN adapter, just try the SolaX Power Modbus custom_component

sanderbaks, 2023-05-07:
Ow, Ok, lets try that!!  I'll let y'all know the results

McLP11, 2023-05-15:
I have the same logger LAN and I cannot as well use this integration, values always unknown. right now the only solution I found, as reported also above, is with SolaX Power modes integration.

adnovea, 2023-05-19:
I have the same problem. I cannot read Modbus from Deye SUN 5K sg03lp1 inverter with a LSE-3 Stick logger. The LAN LSE-3 logger (s/n 21xxxxxxxx) works well with Solarman website but returns incorrect data from local requests based on Solarman V5 protocol. 
For example a request for battery voltage (reg:587) gives me : 
	Tx Packet: a5 1700 1045 b200 <s/n> 02 0000 00000000 00000000 00000000 01 03 024b 0001 <crc16> <crc8>15
	Rx Packet: a5 1700 1045 03 00 bc 02
The return packet is way too short , should have 1015 instead of 1045 and length is wrong.
Does anyone have any idea where to look?

adnovea, 2023-05-28:
McLP11, could you tell more about the use of SolaX Power modes integration with LSE-3 and Deye ? Thanks

McLP11, 2023-05-28:
@adnovea right now SolaX Inverter Modes looks like the only integration I was able to connect with my LSE-3 LAN dongle. It exposes 159 sensors, including also some switches or specific tools for configuration but not tested yet at all. It's pretty stable, I've just noticed that sometimes is not back online when you restart HA, new restart and back again

adnovea, 2023-05-29:
Thanks McLP11 for your reply. I'm trying to understand what is different between SolaX and Solarman to make the LSE-3 working. Is SolaX using a portal and an API ? What I'm looking for, is local communicate with my Inverter using the LSE-3 on my local network only (not using the Solarman portal API). This was possible with the LSW-3. Which version of SolaX have you installed, the one I used with HAC is SolaX Inverter Modbus and expose only 4 sensors.

adnovea, 2023-06-13:
I can communicate with the inverter using the LSE-3 Stick Logger and the Modbus RTU/TCP tester (free) application (https://www.gineers.com/media/downloads/GModbus.zip). The LSE-3 seems to support RAW Modbus RTU/TCP without the V5 Solarman encapsulation. This has to be confirmed. I'm waiting from Solarman feedback about a LSE-3 F/W version supporting V5 protocole.

sanderbaks, 2023-06-13:
This is great to hear. I am following this development!

adnovea, 2023-06-14:
Dear Stephane,
Would you mind to add this protocol to your component in order to support the Ethernet LSE-3 (S/N 21xxxxxxxx) as a selectable option. We will be grateful for this update and again a big thank you for your great work.

gmbh07, 2023-06-20:
I have an zcs azzurro 80 ktl-v3 and the lse-3 Ethernet adapter, but I can't connect. can you help me, I tried your solution but it dosent work.

adnovea, 2023-06-21:
Is your LSE-3 S/N in 21xxxxxxxx ?
Do you have a LSW-3 module and is it working ?

Python indentation is very critical and you better replace Tabs by Spaces.
You can enable the HA debug mode and check the log to see if there is problem with the Solarman modified code

To check if the LSE-3 communication is working  download the GModbus application (https://www.gineers.com/media/downloads/GModbus.zip) and run :
- Enter inverter IP address and port (8899) and press Connect TCP
- Check "Holding", set Address to "1156" (0x484) and Count to "1" to read Grid Frequency x0.01
- Press Read registers
Few seconds later you have the result in the Grid on the right.
In decimal column you should read a value around 5000

If this fails, my mods cannot work for you.

gmbh07, 2023-06-21:
My LSE-4 S/S is in 21xxxxx, I don't have a LSW-3 module because there is no WLAN at this place. I tried the GModbus application with your settings and get something back, look at the picture. 

[image]

adnovea, 2023-06-21:
The GModbus results are encouraging. Your communication to the LSE-3 stick logger is successful.
If you found in the HA log errors such as  "**Querying [0x0400 - 0x042B] failed...**" it's probably a typo issue.
Check the code indentation.

gmbh07, 2023-06-22:
I checked the code and make it all new, now it works for me, thanks for you solution. is there a way to read the modbus register? In the azzurro.yaml there are only two sensors of the strings, and the 80ktl-v3 have some more. By google I didn't found the full modbus register for zcs azzurro

adnovea, 2023-06-23:
Great ! I'm glad the mods works. 
You can read any Azzuro Modbus Register using the GModbus application:
- Connect, select **Holding**, enter start **address** and the **number of register** to read (_max 30 otherwise it could fail repetitively_)

Your string sensors are registers are :

- 0x0584: PV1 Voltage
- 0x0585: PV1 Current
- 0x0586: PV1 Power
- 0x0587: PV2 Voltage
- 0x0588: PV2 Current
- 0x0589: PV2 Power

You can try to read register from 0x058A (1418) to 0x58F (1423) and check if the data pattern looks like the PV1/PV2 ones.
Often, strings data are next to each other.
Sorry, I cannot help you more about the Azzurro.

adnovea, 2023-06-30:
**Modification for LAN LSE-3 Stick Logger module using raw Modbus RTU protocol**
I failed to make the mods working because I have no experience in Python coding.
In _config_flow.py_ I added the following line:
```           
 vol.Optional(CONF_LOOKUP_PROTOCOL, default=data.get(CONF_LOOKUP_PROTOCOL)): vol.In(LOOKUP_PROTOCOL),
```
but I cannot correctly retrieve the protocol radio-button value in the Solarman code.
Could someone help ? Thanks.

[image]

michaelcorradi, 2023-08-06:
I have the same issue. I can't communicate in any way, all parameters remain at "Unknown". I tried to follow different configurations but I can not solve.

My configuration is:
- Inverter "ZCS HYD-6000-HP" with firmware "V40022_2023-05-22".
- Ethernet logger "LSE-3" with firmware "ME_0D_270A_1.09".

Does anyone have an idea how to fix it?

michaelcorradi, 2023-08-06:
> folder and then apply the modifications below using

@adnovea Does this solution work?

adnovea, 2023-08-06:
May I suggest you to follow the steps below:

1. Find the Modbus register list (_Internet search_) corresponding to your Inverter (https://github.com/StephanJoubert/home_assistant_solarman/files/10187824/solarman-ZCS_HYD-6000.txt)
2. Use GModbus utility to test some registers and validate the list.
3. Check if the file **zcs_azzurro-ktl-v3.yaml** has the correct register values. If not, update the YAML file according to your Inverter registers (_you can create a copy of the **zcs_azzurro-ktl-v3.yaml** file into a file named **custom_parameters.yaml** and select this file in the HA Solarman module configuration_). 
4. Modify the Solarman module files **parser.py** and **solarman.py** as I mentioned above (_create backup file to roll back if needed_).

If you succeed in the steps above, you'll be able to retrieve data from your inverter.

_N.B. I you encounter some troubles connecting with GModbus, may be your stick logger F/W need to be updated. You can email the customer service at Solarman customerservice@solarmanpv.com and they will do the update remotely._

see also https://github.com/StephanJoubert/home_assistant_solarman/discussions/148

adnovea, 2023-08-06:
Yes my mods works with Deye and has been used with Azzuro inverter by gmbh07.
Copy the _solarman-ZCS_HYD-6000.txt_ above into _custom_parameters.yaml_, 
apply the modifications,
then select the _custom_parameters.yaml_file in HA Solarman module 
and full reboot HA.

gmbh07, 2023-08-07:
the solution from adnovea for my azzurro and LSE-3 is still working with no problems

michaelcorradi, 2023-09-02:
@adnovea with your suggestion my configuration is working.

My configuration is:
- Inverter "ZCS HYD-6000-HP" with firmware "V40022_2023-05-22".
- Ethernet logger "LSE-3" with firmware "ME_0D_270A_1.09" and SN "21xxxxxxxx".

I'm using the configuration "zcs_azzurro-ktl-v3.yaml" is the only one that works, but I see only few values updated.

What is the best configuration I can use?
Since your code works, can't the developer @StephanJoubert upload it to the official library?

adnovea, 2023-09-02:
Hi Michael,

Glad to see that the mods is working with your config.
I hope @**StephanJoubert** will update the code for supporting the LSE-3.

Have you tried to download the configuration file at https://github.com/StephanJoubert/home_assistant_solarman/files/10187824/solarman-ZCS_HYD-6000.txt ?
Did it expose more values ?

michaelcorradi, 2023-09-02:
I tried to put the content fo your txt file in "custom_parameters.yaml" (27 entities) but the communication does't work.
I wait for some minute but it remain Disconnected.

adnovea, 2023-09-02:
Once you have :
1-  created the "_custom_parameters.yaml_"
2- fully restarted HA
3- cleared the browser 
4- and re-logging to HA, 
you should have 27 entities exposed by the integration.

If this still fails, try to communicate with you inverter using **GModbus** (_see above how to run this utility_). 
**Attention**: register values shall be entered as decimal (_not Hex_) values.

maartenba, 2023-09-05:
For ME3000-SP and the LSE-3, using the sofar_wifikit preset and changes by @adnovea (https://github.com/StephanJoubert/home_assistant_solarman/issues/247#issuecomment-1591817882) worked wonders.

maartenba, 2023-09-06:
Change seems to be incompatible with 1.5.0 because of a switch to pysolarmanv5. @adnovea not sure if you know how to work around that? I'm happy to open a pull request with a enable/disable V5 encapsulation toggle, though for 1.5.0 I'm not sure how to disable V5 anymore.

adnovea, 2023-09-06:
Hi maartenba,

I made the mods for 1.0.0.
**Basically the LSE-3 support a RAW Modbus protocol.**
To support the LSE-3 module, the V5 solarman protocol that encapsulate the RAW Modbus data just need to be optional.
My mods was only commenting the code line of the encapsulation.

I'm sure Stephan know how to do this but I'll be please to help when time permits.

Mocky158, 2023-11-19:
I would like to ask if someone would be willing to upload modified files here for use by LSE, somehow I can't do it. Thank you very much .

adnovea, 2023-11-20:
Hi Mocky158, if you can wait a about a week, I'm creating a repo (_solarman_lse_) with a fork from the nice Stephan Add-on to support the LSE stick loggers. The purpose of this repo is mainly to share a quick solution for those with LSE Stick Logger. It is compatible with the inverter_definitions and provides a more flexible solution for Deye Sun 5K with possibility to write registers. For other inverters, owner should dig in the inverter Modbus register documentation to find the corresponding codes and values (_BE CAREFUL because this can definitely damage the inverter_).

Niklas1E, 2024-01-19:
@adnovea 

You describe here (https://github.com/StephanJoubert/home_assistant_solarman/issues/247#issuecomment-1590665761
) that you adapt two files (parser.py and solarman.py) and the LAN dongle then works on the DEYE SUN?
Maybe I'm being very stupid, but I can't find the values in the two files.
My dad has the DEYE SUN and I want to help him integrate it into HomeAssitant.

THX

adnovea, 2024-01-20:
Niklas1E, the last version of Solarman use the Solarman library instead of the files mentioned above.
That's the reason why I had to fork this project into the repo [_solarman_lse_](https://github.com/adnovea/solarman_lse).

Niklas1E, 2024-01-20:
Thank you very much!Can I damage anything on the inverter with the add-on?Or does it only read data or does it also write data?Sorry, I'm an amateur.Am 20.01.24 um 08:23 schrieb AdNovea®
                
            
            
                
                    Von: "AdNovea®" ***@***.***>Datum: 20. Januar 2024An: "StephanJoubert/home_assistant_solarman" ***@***.***>Cc: "Comment" ***@***.***>,"Niklas1E" ***@***.***>Betreff: Re: [StephanJoubert/home_assistant_solarman] Switched from wireless lan logger LSW-3 to cable logger LSE-3 (Issue #247)
                
                
Niklas1E, the last version of Solarman use the Solarman library instead of the files mentioned above.
That's the reason why I had to fork this project into the repo solarman_lse.

adnovea, 2024-01-20:
There is always a risk when you don't perfectly understand what you do.
Read carefully all the instructions. This is not a "plug&play" module. It has not been tested with many inverters nor fully debugged.
I just share what I'm currently using for my Deye Sun 5K but you must understand what you are doing.
As long as you avoid writing to the inverter registers, the risk is limited.
Do not use switch entities to prevent sending unexpected commands and keep only read registers.

Imanol82, 2024-04-01:
Hello @adnovea . I'm in the same situation.
Deye Sun 6K, with LSE-3 SN: 21*******

1- created the "custom_parameters.yaml"
2- IP ans SN:
[image]
3- Configuration in HA:
[image]
4- Result:
[image]

Why? What I'm doing wrong?
Thank you very much!

adnovea, 2024-04-02:
Imanol82 did you read the whole post ?
I explain above why the Stepan module does not work with LSE module and what can be done to solve the problem.
The current version cannot be modified for LSE and you need to tune the add-on with my solarma_lse fork.

Imanol82, 2024-04-02:
Thank you very much!

I'm very new in HA... I'll try mod.

:)

70sxorpio, 2024-05-09:
Hello. I have an Afore BNT004KTL inverter with LSE-3 logger serial number 21xxxxxxxx and no configuration works in homeassistant. Does anyone have this inverter connected? Please help.

MartinNeurol, 2024-05-27:
> There is always a risk when you don't perfectly understand what you do. Read carefully all the instructions. This is not a "plug&play" module. It has not been tested with many inverters nor fully debugged. I just share what I'm currently using for my Deye Sun 5K but you must understand what you are doing. As long as you avoid writing to the inverter registers, the risk is limited. Do not use "switch/select" entities to prevent sending unexpected commands and keep only "sensor" registers.

Hi! I am using the Solarman LSE for my SUN-10K-SG04LP3-EU. Everytime I set up the interation, in less then 24 HRS, my inverter falls into COM error and has to be restarted. That never happens when the integration is off. As I am using LSE version and my neigbor literally has the same solution, but with standard Solarman integration and wireless logger working just fine, I am just asking  -  where to look, who to ask? Or ditch it? Many thanks, Martin