StephanJoubert/home_assistant_solarman #249: how to reset daily production

Bugahans, 2023-03-21:
The daily production counter is not reset automatically. It shows the same value like the Total Production.
I read already that you ahve to reset it yourself if you dont use the Solarman app (cloud). Is it possible to send the required command to the register to reset the daily production?

haraldfunk, 2023-03-21:
Have spottet the same behavior after I cut off the inverter from internet access. Any way to send a reset signal from HA would be highly welcome for me as well.

r3pl1x, 2023-03-21:
As a workaround you could use this integration:
https://www.home-assistant.io/integrations/utility_meter/

JamesBHelix, 2023-03-23:
I use a Deye600 here and try to run it without solarman as well. 
Current status: 
- Beginner :)
- Hass.io on RPI 4
- All data, except for the day counter is working. 

Have blocked the port of the application on the firewall, so the inverter still gets its time from cn.ntp.org.cn. ( can be read out via AT-Command)
Modbus: In register 16 is normally the last query values date/time .
The next morning starts with:0.
If you set a date/time here, the daily production counter starts with 0 again. 

Ideally it would be, with each inquiry by ha-solarman, the values date/time of the inquiry in register 16 to set. 

This is still missing here. 
Does anyone have an idea?

JamesBHelix, 2023-03-24:
I use a Deye600 here and try to run it without solarman as well.
Current status:

HA Beginner :)
Hass.io on RPI 4
All data, except for the day counter is working.
Have blocked the port of the application on the firewall, so the inverter still gets its time from cn.ntp.org.cn. ( can be read out via AT-Command)
Modbus: In register 16 is normally the last query values date/time .
The next morning starts with:0.
If you set a date/time here, the day counter starts with 0 again.

Ideally it would be, with each inquiry by ha-solarman, the values date/time of the inquiry in register 16 to set.

This is still missing here.
Does anyone have an idea?

haraldfunk, 2023-04-30:
I tried to update the registers 22/23/24 with the write_register service introduced by solarman integration in HA. Unfortunately it seems that nothing happens when calling the service (e.g. with values 0x2304 / 0x3020 / 0x1006 for 30th of April 2023 20:10:06). The content of the registers stays the same.
Also it's not clear to me, which of my 2 inverters is called by this service as there is no way to choose between them. Either way non of them accepts the data and is changing the values.

By using the python script of https://github.com/jedie/inverter-connect it works, so in general it should be feasible.

Any advice from @StephanJoubert on what I am doing wrong? Is this write_register service tested and working for you?

JamesBHelix, 2023-05-31:
I use  the program main under linux. 
main.exe under windows doesn´t support write command. 

command:  ./main -t  IP_address:port  -xmbw 001600030617051F053000
Port:48899   (most)                                                    Register x16000306   Start 16, 3 Register, 6 Length
                                                                       Date:  170501F050300    230523  05:48:00

prankousky, 2023-07-15:
> By using the python script of https://github.com/jedie/inverter-connect it works, so in general it should be feasible.

@haraldfunk how do you use the python script to reset the daily counter? I installed it and set it up. The connection is established, i.e. I can run `./cli.py inverter-version` and receive valid output. But what do I need to do in order to reset daily production?