StephanJoubert/home_assistant_solarman #252: (SyncWorker_3) [custom_components.solarman.solarman] Querying inverter 2208xxxxxx at x.x.x.x:8899 failed on connection start with exception [ConnectionRefusedError]

audio-catalyst, 2023-03-23:
Anyone got any idea?
ip adress and serial are 100% correct and ip is pingable and web interface is reachable

Device Information
Serial Number:
---------------------
Register Key:
---------------------
Hardware Version:
AEW2-0005-03
Software Version:
4790123201R

Cloud Information
Connection Status:
OK

thorsten-l, 2023-04-07:
Got the same error. (deye sun600g3-eu-230)

wax911, 2023-04-08:
I never used to have this issue before but lately I've been I've been getting timeouts

DominicStenzel, 2023-04-29:
Same here, never had an issue before.

This is the error:

This error originated from a custom integration.

Logger: custom_components.solarman.solarman
Source: custom_components/solarman/solarman.py:173
Integration: Solarman (documentation, issues)
First occurred: 12:45:04 (7 occurrences)
Last logged: 12:51:04

Querying inverter xxxxxxxxx at 192.168.x.xxx:8899 failed on connection start with exception [OSError]

wax911, 2023-04-30:
Hopefully this will help, debug the issue further:

```log
Logger: homeassistant
Source: custom_components/solarman/__init__.py:35
Integration: Solarman (documentation, issues)
First occurred: 19:02:26 (2 occurrences)
Last logged: 19:03:08

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solarman/__init__.py", line 35, in update_listener
    hass.data[DOMAIN][entry.entry_id].config(entry)
KeyError: 'solarman'
```