StephanJoubert/home_assistant_solarman #258: 27**** inverter

nemenos, 2023-04-02:
Hello,
I cannot make it work with my 27**** serial number inverter  (I have a ZCS Azzurro HYD-ZSS 5000).
It works using non hybrid configuration (zcs_azzurro-ktl-v3.yaml) but it misses lots of sensors, 
with sofar_hyd3k-6k-es.yaml it doesn't work at all
thanks in advance

nemenos, 2023-04-05:
I just think it's a matter of different registers. Anybody knows how to find documentation for reading correct registers for this inverter?
Thank you

nemenos, 2023-04-24:
#279

vinceg0267, 2023-04-30:
Ciao, io ho il protocollo sofar con i registri per gli inverter HYD xxxx EP che sono uguali ai ZCS HP. Tu sei capace a creare un yaml da questi registri?
[Protocol_Sofar-EP.xlsx](https://github.com/StephanJoubert/home_assistant_solarman/files/11360953/Protocol_Sofar-EP.xlsx)

vinceg0267, 2023-04-30:
> I just think it's a matter of different registers. Anybody knows how to find documentation for reading correct registers for this inverter? Thank you

[Protocol_Sofar-EP.xlsx](https://github.com/StephanJoubert/home_assistant_solarman/files/11360963/Protocol_Sofar-EP.xlsx)

nemenos, 2023-05-01:
Ciao, è solo questione di correggere i registri. (Solo se sono diversi, magari domani faccio un controllo e potresti usare anche direttamente la mia configurazione) .
Il numero (o i numeri) del registro lo trovi sul foglio "descrizione dell'indirizzo" e lo sostituisci facilmente, insieme al valore che trovi sullo "scale" e sull'unità di misura. Anche il mode è importante in base al tipo di da dato (U16) etc, e la classe. Ma andando un po' a intuito rispetto alle altre configurazioni si capisce abbastanza brne
Comunque considera che non conosco python e dopo qualche prova sono riuscito, anche velocemente...

vinceg0267, 2023-05-02:
Quindi è sufficiente editare lo yaml in questione e modificare i registri necessari?

nemenos, 2023-05-02:
Si, se aggiungi uno yaml nuovo, per farglielo vedere va anche aggiunto sul const.py (se ricordo bene).
comunque se guardi il mio pull request cmq si capisce bene.
Adesso che ci penso, oltre a modificare i registri va anche specificato in cima al file quali sono i registri richiesti

vinceg0267, 2023-05-02:
Ho aggiunto il tuo yaml nella cartella inverter definitions e aggiunto il nome in const.py. Non vedo più le entità.

vinceg0267, 2023-05-02:
Ah, dimenticavo, io ho creato un gruppo su facebook per i possessori di inverter zcs, dove condividiamo problemi e soluzioni
https://www.facebook.com/groups/509947030592700

nemenos, 2023-05-02:
> Ho aggiunto il tuo yaml nella cartella inverter definitions e aggiunto il nome in const.py. Non vedo più le entità.

mah, mi sembra strano... io grossi problemi non ne ho avuti.

vinceg0267, 2023-05-02:
sicuramente sbaglio qualcosa io, non capisco un tubo di queste cose. Ho copiato uno yaml sul desktop, ho cancellato il codice esistente e ho copiato il tuo, ho salvato con il nuovo nome e ho aggiunto il nome in const.py.
Poi su impostazioni, dispositivi, ho selezionato solarman e configura e ho cambiato lo yaml in basso. Ho riavviato HA e mi da le entità non disponibili.
benvenuto nel gruppo

eamarchese68, 2023-06-06:
Thanks to the info in this group, some searches in Google and some tests, I could generate a working definition file starting from the 3PH version sofar_g3hyd.yaml.
Here is the code.

requests:
  - start: 0x0404
    end: 0x0428
    mb_functioncode: 0x03
  - start: 0x0484
    end: 0x04AF
    mb_functioncode: 0x03
# off - grid info
#  - start: 0x0504
#    end: 0x051F
#    mb_functioncode: 0x03
  - start: 0x0584
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0604
    end: 0x060A # end of first battery after this continue battery pack 2,3,4
    mb_functioncode: 0x03
  - start: 0x0684
    end: 0x069B
    mb_functioncode: 0x03
parameters:
  - group: Inverter
    items:
      - name: "Inverter status"
        class: ""
        state_class: "measurement"
        uom: ""
        scale: 1
        rule: 1
        registers: [ 0x0404 ]
        isstr: true
        lookup:
          - key: 0
            value: "waiting"
          - key: 1
            value: "detection"
          - key: 2
            value: "grid-connected"
          - key: 3
            value: "emergency power supply"
          - key: 4
            value: "recoverable fault"
          - key: 5
            value: "permanent fault"
          - key: 6
            value: "upgrade"
          - key: 7
            value: "self-charging"
        icon: 'mdi:wrench'
      - name: "Ambient temperature"
        class: "temperature"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x0418 ]
        icon: 'mdi:thermometer'
      - name: "Radiator temperature"
        class: "temperature"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x041A ]
        icon: 'mdi:thermometer'
      - name: "Daily Generation Time"
        class: "time"
        uom: "min"
        scale: 1
[... 1038 lines cut ...]
            value: 'ID180 BMS low temperature alarm'
          - key: 16
            value: 'ID181 BMS charge/discharge overload alarm'
          - key: 32
            value: 'ID182 BMS short circuit alarm'
          - key: 64
            value: 'ID183 BMS version inconsistency'
          - key: 128
            value: 'ID184 BMS CAN version inconsistency'
          - key: 256
            value: 'ID185 BMS CAN version is too low'
          - key: 512
            value: 'ID186 '
          - key: 1024
            value: 'ID187 '
          - key: 2048
            value: 'ID188 '
          - key: 4096
            value: 'ID189 Arc device communication failure'
          - key: 8192
            value: 'ID190 DC arc alarm fault'
          - key: 16384
            value: 'ID191 PID repair failed'
          - key: 32768
            value: 'ID192 PLC module heartbeat loss'

eamarchese68, 2023-06-06:
Starting from the comments in this issue, some google searches and some test, I could generate a working definition file for the 1PH version starting from the 3PH version sofar_g3hyd.yaml
Here is the code.

```

requests:
  - start: 0x0404
    end: 0x0428
    mb_functioncode: 0x03
  - start: 0x0484
    end: 0x04AF
    mb_functioncode: 0x03
# off - grid info
#  - start: 0x0504
#    end: 0x051F
#    mb_functioncode: 0x03
  - start: 0x0584
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0604
    end: 0x060A # end of first battery after this continue battery pack 2,3,4
    mb_functioncode: 0x03
  - start: 0x0684
    end: 0x069B
    mb_functioncode: 0x03
parameters:
  - group: Inverter
    items:
      - name: "Inverter status"
        class: ""
        state_class: "measurement"
        uom: ""
        scale: 1
        rule: 1
        registers: [ 0x0404 ]
        isstr: true
        lookup:
          - key: 0
            value: "waiting"
          - key: 1
            value: "detection"
          - key: 2
            value: "grid-connected"
          - key: 3
            value: "emergency power supply"
          - key: 4
            value: "recoverable fault"
          - key: 5
            value: "permanent fault"
          - key: 6
            value: "upgrade"
          - key: 7
            value: "self-charging"
        icon: 'mdi:wrench'
      - name: "Ambient temperature"
        class: "temperature"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x0418 ]
        icon: 'mdi:thermometer'
      - name: "Radiator temperature"
        class: "temperature"
        uom: "°C"
        scale: 1
        rule: 2
        registers: [ 0x041A ]
        icon: 'mdi:thermometer'
      - name: "Daily Generation Time"
        class: "time"
[... 1042 lines cut ...]
            value: 'ID181 BMS charge/discharge overload alarm'
          - key: 32
            value: 'ID182 BMS short circuit alarm'
          - key: 64
            value: 'ID183 BMS version inconsistency'
          - key: 128
            value: 'ID184 BMS CAN version inconsistency'
          - key: 256
            value: 'ID185 BMS CAN version is too low'
          - key: 512
            value: 'ID186 '
          - key: 1024
            value: 'ID187 '
          - key: 2048
            value: 'ID188 '
          - key: 4096
            value: 'ID189 Arc device communication failure'
          - key: 8192
            value: 'ID190 DC arc alarm fault'
          - key: 16384
            value: 'ID191 PID repair failed'
          - key: 32768
            value: 'ID192 PLC module heartbeat loss'

```