StephanJoubert/home_assistant_solarman #263: An ethernet connection with this integration does not work for me

sanderbaks, 2023-04-09:
Hi!

My inverter is not in range of my wifi, because I have my PV panels outside in the field. So I use the ethernet dongle. 
FW version of the dongle is ME_0C_270A_1.05
Using a portscanner I can see that port 8899 is open.
My iphone solarman app is working.
The serial number start with 21, indicating a v5 data configuration.
In HAS I use the sofar_lsw3.yaml file for the integration.

However, there is no connection, and all the entities remain unknown.

I am not interested in getting the API option working. I want to understand what is going on here :)

Any tips are welcome!

spanxter, 2023-04-29:
I have the exact same problem…

sanderbaks, 2023-05-02:
Anyone ?  Or any idea how to troubleshoot this ?

McLP11, 2023-05-12:
same for me, no chance to have data available...

mcasteels, 2023-05-26:
I had exactly the same issues.  My Sofar is on Ethernet since I rather have devices wired instead of wifi.  Always had isues to get it working.  Thanks to an other users I tried the Sofar_Wifikit and all worked.

Adonaisdererste, 2023-05-28:
SUN-12K-SG04LP3-EU
LAN Dongle : Stick Logger(Ethernet) Model LSE-3

I have the same problem ,
i get no data.

Please find a solution,because i don't want to use a WIFI connection in a public place.