StephanJoubert/home_assistant_solarman #266: deye 6k g03 returning all values unknows

denveronly, 2023-04-12:
Hi there.
Ive installed the integration and tried to make the connection.
Homeassistant makes a device with 50 entities that are all Unknown values. 
There are no bad logs in Homeassistant.

Device serial number2742XXXX91
Firmware versionLSW3_15_FFFF_1.0.92R

PV-Joe, 2023-04-12:
What definition did you try?

[Solarman definitions](https://github.com/StephanJoubert/home_assistant_solarman/tree/main/custom_components/solarman/inverter_definitions)

denveronly, 2023-04-12:
Well i tried all of them. Mine should be just deye_hybrid. But it was not connecting
It only got connected in 0.0.0.0 mode 
In the device settings i see deye_hybrid

i did a homeassistant full restart, before that i was adding it without restarts

denveronly, 2023-04-12:
[image]
Alao strange that i have l2 l3.
My inverter is single phase