StephanJoubert/home_assistant_solarman #267: HA & Deye 5kw Inverter with Stick logger model LSW-3-C

Kobus4ster, 2023-04-15:
I am trying to use the Solarman integration to Deye 5kw Inverter with Stick logger model LSW-3-C
[image]
From below "Unavailable" on all Sensors, it seems that the connection cannot be made to the IP (in the above)
[image]
I have set the configuration as follows 
[image]
I added the the sensor in the configuration.yaml as
[image]

I have two topics that expect can be the source of my problem:
1) Since the serial number 2719597711 is not in the V5-range, this could be a potential problem, but not sure what exactly need to change, should I need V4 instead.

2) I also have an issue to access the Logger IP 192.168.0.100 from my browser (I tried Brave, Chrome & Edge) - all give the same error: "This site can’t be reached192.168.0.100 refused to connect.". I can see the IP in my DHCP listing and I can also PING the IP with no problem. Trying to resolve this issue, I tried:
- Firewall settings (and even switched it off)
- Antivirus settings (and also set it off)
- tries another PC on my network
but the problem remain

PS: I was able to open the logger when I connected directly to the loggers AP_* WiFi network that it created. I am not sure what the "Remote server B = not connected" means ?
[image]

denveronly, 2023-04-16:
try connecting to 192.168.0.102 that is your logger IP that is seen on the last screenshot
i have the same V5 with 27xxxxx
to make the connection i used 0.0.0.0 as an ip in the HA integration section and device id is 0
i got it connected but it works not as good as expected.
Battery dischage for example worked for a day registering perfectly, next day i got it not working with last message 10 hours ago (battery was discharging during the daytime)

Kobus4ster, 2023-04-18:
Thanks for you help @denveronly - unfortunately I can also not access or PING the IP 192.168.0.102.
I have also tried the 0.0.0.0 IP and device id is 0 in the HA, but still get the "unavailable" response. It's really like the logger is not available on my network, but it's IP IP 192.168.0.100 is in the DHCP list on my network. You gave me hope that the connection is possible. Thanks

denveronly, 2023-04-18:
> Thanks for you help @denveronly - unfortunately I can also not access or PING the IP 192.168.0.102.
> I have also tried the 0.0.0.0 IP and device id is 0 in the HA, but still get the "unavailable" [...]

I think that it is not the stick issue, it is your network problem the ip .100 or .102 is already registered by other device
Maybe try specifying in Stick settings not a fhcp connection, but a static ip with
192.168.0.1 router
192.168.0.254 or 253 which absolutely not taken, please scan as Stick ip
255.255.255.0 as subnet mask

after apllying settings try scan one more time if the device had appeared on network
Inverter sticks are too dumb to fail to connect to network

Kobus4ster, 2023-04-23:
> 
Yes! it's working now - Thanks @denveronly for your advice. I changed the network settings as you suggested and then I could see the logger on my home network with the static IP. I plugged the IP into the HA configuration.yaml file. However, I had to delete and add the Solarman add-in again. Then the Solarman "device" in HA connected and started to show the values from the logger. After about an hour I started to see data on the Energy dashboard. Thanks a mil for your assistance.

closed 2023-04-23 by Kobus4ster

matauanu, 2024-07-01:
Hello! The datas are live ? No delay, directly from wifi stick or are through solar man servers?