StephanJoubert/home_assistant_solarman #276: Deye SUN 800 - Power per MPPT

WouterJN, 2023-04-22:
In the solar man application I can see the power (Watt) per MPPT for my SUN800G3. This information is not shown in Home Assistant. Could power per MPPT be added?

DHLF, 2023-04-23:
The SUN800G3 with two mppt transmits total power(watts) for BOTH(PV1+PV2) combined.
As the data for each PV module is given in amps and voltage you can calculate it for yourself by multiplying both values:
{{ states('sensor.deye_pv1_voltage') | float * states('sensor.deye_pv1_current') | float }}
{{ states('sensor.deye_pv2_voltage') | float * states('sensor.deye_pv2_current') | float }}
and of course for those with 4 mppts:
{{ states('sensor.deye_pv3_voltage') | float * states('sensor.deye_pv3_current') | float }}
{{ states('sensor.deye_pv4_voltage') | float * states('sensor.deye_pv4_current') | float }}

You have to change the naming for your setup so instead of deye_ you have to use whatever you typed when installing.

I am looking for a proper solution for the 4mppt(2000G3) myself.. as this may be the proper calculation but hence.. i am unable to create new "identities" like deye_pv1_power YET.
Putting this stuff in the configuration.yaml did not help as the new "identities" cant be seen anywhere -.-

```
solarman:
  - platform: template
    sensors:
      deye_pv1_power:
      unit_of_measurement: W
      device_class: power
      value_template: {{ states('sensor.deye_pv1_voltage') | float * states('sensor.deye_pv1_current') | float }}
```

DHLF, 2023-05-07:
Crawling around I *may* have found a solution:
Putting this into the /config/configuration.yaml:
```
template:
  - sensor:
      - name: deye_2k_pv1_wattz
        unit_of_measurement: "W"
        state: "{{ (states('sensor.deye_2k_pv1_voltage') | float) * (states('sensor.deye_2k_pv1_current') | float) }}"
        icon: "mdi:solar-power-variant-outline"
      - name: deye_2k_pv2_wattz
        unit_of_measurement: "W"
        state: "{{ (states('sensor.deye_2k_pv2_voltage') | float) * (states('sensor.deye_2k_pv2_current') | float) }}"
        icon: "mdi:solar-power-variant-outline"
      - name: deye_2k_pv3_wattz
        unit_of_measurement: "W"
        state: "{{ (states('sensor.deye_2k_pv3_voltage') | float) * (states('sensor.deye_2k_pv3_current') | float) }}"
        icon: "mdi:solar-power-variant-outline"
      - name: deye_2k_pv4_wattz
        unit_of_measurement: "W"
        state: "{{ (states('sensor.deye_2k_pv4_voltage') | float) * (states('sensor.deye_2k_pv4_current') | float) }}"
        icon: "mdi:solar-power-variant-outline"
```
The **-name: deye_2k_pv1_wattz** will be the entity name which should be usable in HA. You may change it like you wish, this example is my definition to divert between different installed microinverters.

If you installed solarman WITHOUT changing the prefix the "states" will be known as "sensor.**solarman**_pv1_voltage" and "sensor.**solarman**_pv1_current". Change it to your setup.

Some things to keep in mind:
1.) you MUST restart HA. Just changing the configuration.yaml will do nothing otherwise..
2.) If the sun is down so the values are not valid and.. error will be shown as the value "undefined" is no float :D 
 You may use a conditional Card using **sensor_solarman_status_connection** comparing to **connected** to hide error cards.
3.) If you got only 2mppts dont use the last two entries as the entities will not be found and error will be shown -.-

Disclaimer: I have not yet tested this because its dark outside now -.- But its the first time HA accepted this stuff in the configuration :)

DHLF, 2023-05-07:
Another missing part(dunno if this is the correct way.. anyway it works..):
The new identities will not be usable in "type: history-graph"-cards.

To let them be known to HA you will have to add another line **state_class:"measurement"** to each entity:
```
      - name: deye_2k_pv1_wattz
        unit_of_measurement: "W"
        state: "{{ (states('sensor.deye_2k_pv1_voltage') | float) * (states('sensor.deye_2k_pv1_current') | float) }}"
        state_class: "measurement"
        icon: "mdi:solar-power-variant-outline"

```
After a restart of HA it may took around 10 minutes+ until the new entities show  up

awalon, 2023-11-01:
Thanks, just added 2 PVs based on this solution with some error handling (0 value, if disconneced):
```
sensor:
  - platform: template
    sensors:
        calc_deye_pv1_power:
          friendly_name: "Deye PV1 Power"
          unique_id: calc_deye_pv1_power
          unit_of_measurement: "W"
          value_template: >-
            {% if not is_state('sensor.deye_pv1_voltage', ['unknown', 'unavailable'] )
                 and is_number(states('sensor.deye_pv1_voltage')) -%}
               {{ states('sensor.deye_pv1_voltage') | float * states('sensor.deye_pv1_current') | float }}
            {%- else -%}
               0
            {%- endif %}
          device_class: "power"
          icon_template: "mdi:solar-power-variant-outline"
        calc_deye_pv2_power:
          friendly_name: "Deye PV2 Power"
          unique_id: calc_deye_pv2_power
          unit_of_measurement: "W"
          value_template: >-
            {% if not is_state('sensor.deye_pv2_voltage', ['unknown', 'unavailable'] )
                 and is_number(states('sensor.deye_pv2_voltage')) -%}
               {{ states('sensor.deye_pv2_voltage') | float * states('sensor.deye_pv2_current') | float }}
            {%- else -%}
               0
            {%- endif %}
          device_class: "power"
          icon_template: "mdi:solar-power-variant-outline"
```