StephanJoubert/home_assistant_solarman #277: deye_4mptt.yaml PV1,PV3 and PV4 current not usable in statistics

DHLF, 2023-04-23:
While the entities for PV1,PV2,PV3,PV4(current) are displayed in the default view(with proper values)
only PV2 current can be used in statistic-graph(the other three can't be found).

A quick look into deye_4mptt.yaml showed that PV1,PV3 and PV4 CURRENT are missing the `state_class: "measurement"`
Adding this line properly and restarting HA did not help(well it did but it took some time :) ).
The before missing identities can NOW be selected for use in statistic-graph cards.

Hint: Be aware that it will take some time until the values finally show up in the statistical-graph card.

Kolbi, 2023-05-23:
Should be fixed with: https://github.com/StephanJoubert/home_assistant_solarman/pull/304

Sycend, 2024-02-24:
Still have this issue on PV1 with the deye_string.yaml