StephanJoubert/home_assistant_solarman #287: Running Status "Fault" all the time with deye_2mppt.yaml

SmoothMC, 2023-05-04:
Hey,

I used the deye_2mppt.yaml at the moment but with deye_4mppt.yaml it is the same

Inverter: DEYE SUN1000G3
Firmware: MW3_16U_5406_1.53

With those definitions, the "Running Status" is "Fault" all the Time. In the WebUI from the DEYE itself there is no issue displayed.

Maybe there is some mismatch in the yaml? I don't know. Is there anyone else with the same Display issue?

  - group: Inverter
    items:
    - name: "Running Status"
      class: ""
      state_class: ""
      uom: ""
      scale: 1
      rule: 1
      registers: [0x003B]
      isstr: true
      lookup:
      - key: 0
        value: "Stand-by"
      - key: 1
        value: "Self-check"
      - key: 2
        value: "Normal"
      - key: 3
        value: "Warning"
      - key: 4
        value: "Fault"
      icon: 'mdi:home-lightning-bolt'

Thx for your work and help

PV-Joe, 2023-05-06:
The “Running Status” entity is missing for the Deye Hybrid inverter:

[Deye 8K Hybrid: Solarman Running Status is missing after updating from 1.4.1 to 1.4.3 #261](https://github.com/StephanJoubert/home_assistant_solarman/issues/261)

DHLF, 2023-05-06:
Deye-2mppt.yaml (deye Sun600G3EU) Version 22.02.2023
The "older" Version did not show any fault, but now state is shown as fault all the time.

Comparing to the deye_4mppt.yaml running "on" a deye SUN 2000 G3 EU there is no problem as "normal" is shown as state.
Both deyes are not in the cloud but locked in the local net.

Either the Sun600 is damaged(but there is no error code shown) or both devices are different somehow.

vordenken, 2023-05-20:
Get the same message on my sun600g3eu, no cloud, addon version 1.4.1

b-reich, 2023-06-29:
Mine SUN-M80G3-EU - main branch and 2mppt.yaml also reporting "fault"

G0ldmember, 2023-10-08:
I have the same issue with an Deye M60, but I don't think it is a YAML problem, as the Solarman web site also shows that error in the "running state" section of the inverter. So the interpretation is correct, still I'm wondering what error this is.