StephanJoubert/home_assistant_solarman #288: Home Assistant 2023.5.1 broke Solarman Integration

PV-Joe, 2023-05-05:
I updated to the latest Home Assistant 2023.5.1 this morning and it broke the Solarman Integration 1.4.1 for my Deye 8K Hybrid with the following error:

2023-05-05 06:13:34.327 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solarman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-05-05 06:13:47.649 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Deye 192.168.178.89 for solarman
File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry
2023-05-05 06:14:22.244 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.solarman_daily_energy_bought
2023-05-05 06:14:22.249 WARNING (MainThread) [homeassistant.components.energy.sensor] Found unexpected state_class None for sensor.solarman_daily_energy_sold

Does anyone know how to fix that?
Any suggestions much appreciated!

Joe

mathiaswagner, 2023-05-05:
0cd2e8d fixed that.

PV-Joe, 2023-05-05:
Updating to Solarman Integration 1.4.3 fixed the problem with Home Assistant 2023.5.1.

The only entity that is missing since updating is the “Running Status” but I can live with that for now. It was working in Solarman 1.4.1 but the code was removed from the YAML. Adding it back into the YAML broke the integration.

I hope that it can be fixed in a future release.

vordenken, 2023-05-24:
I don't know if this is already tracked in another issue but (at least for me) only version 1.4.1 works with my deye sun600g3.
I cannot get 1.4.3 to work with that. So updating to 1.4.3 is no option for me atm...

wolak27, 2023-05-26:
Hello I have simlar issue, I've installed today 2023.5.4 from 2023.4...
Everything worked good before this update :( 

 
```
Logger: homeassistant.config_entries
Source: custom_components/solarman/__init__.py:18
Integration: Solarman ([documentation](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md), [issues](https://github.com/StephanJoubert/home_assistant_solarman/issues))
First occurred: 14:32:57 (3 occurrences)
Last logged: 14:47:28

Error setting up entry Fotowoltaika for solarman
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solarman/__init__.py", line 18, in async_setup_entry
    hass.config_entries.async_setup_platforms(entry, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'
```

Otanar1, 2023-06-20:
> I don't know if this is already tracked in another issue but (at least for me) only version 1.4.1 works with my deye sun600g3. I cannot get 1.4.3 to work with that. So updating to 1.4.3 is no option for me atm...

Easy to solve, just go to:
`custom_components/solarman/__init__.py`

Change the line:
`hass.config_entries.async_setup_platforms(entry, PLATFORMS)`

For:
`await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)`

It will look like this image:
[image]

Otanar1, 2023-06-20:
> Hello I have simlar issue, I've installed today 2023.5.4 from 2023.4... Everything worked good before this update :(
>
> ```
> Logger: homeassistant.config_entries
> Source: [...]

see the solution above