StephanJoubert/home_assistant_solarman #293: LSW3_15_1E03_1.31

ricoroodenburg, 2023-05-07:
Hi,

Have anyone luck with connecting to a inverter with23x serial number and firmware version LSW3_15_1E03_1.31?

I don’t get any data from the logger.

teppic01, 2023-05-09:
Hi,
same for me, fw MW3_16U_5406_1.57. By the installation there is no deye_2mppt.yaml to choose.
With the deye_hybrid.yaml it does not work.
There is nothing created in the configuration.yaml by installation. Manual integration does not work to.

sanderbaks, 2023-05-09:
Same here, the LSW versions dont seem to work.. No idea why

teppic01, 2023-05-10:
Solution for me :
First copy deye_2mppt.yaml from website. Paste into new file "custom_parameters.yaml" into directory.
Reconfigurate the solarman intergration and choose this file. Next day it worked. (maybe after restart of deye and ha?)
good luck

ricoroodenburg, 2023-05-10:
Hi,

Doesn’t work for me. I have reboot both the inverter and home assistant instance.

Greetings,
Rico

ricoroodenburg, 2023-05-11:
> You have to deinstall the integration in HACS, after that install the Version "MAIN" (choose in dropdown, the very first option)
> This will install the 1.4.3
>
> The version options above "main" just install crap

I’ve tried this without any luck…

thhomesystem, 2023-05-28:
> Solution for me : First copy deye_2mppt.yaml from website. Paste into new file "custom_parameters.yaml" into directory. Reconfigurate the solarman intergration and choose this file. Next day it worked. (maybe after restart of deye and ha?) good luck

Where can we download this file ?

teppic01, 2023-06-01:
Sorry for the long time to answer...
Just on this site
https://github.com/StephanJoubert/home_assistant_solarman/tree/main/custom_components/solarman/inverter_definitions

ricoroodenburg, 2023-09-11:
Nobody?