StephanJoubert/home_assistant_solarman #298: Show state of charge?

membersound, 2023-05-17:
Isn't it possible to show the present state of charge. Maybe in %, or even in kWh?
I don't mean the `sensor.solarman_battery_power` which shows currently used power.
I mean what's left inside the battery. Is there no sensor for it?

PV-Joe, 2023-05-17:
> Isn't it possible to show the present state of charge. Maybe in %, or even in kWh? I don't mean the `sensor.solarman_battery_power` which shows currently used power. I mean what's left inside the battery. Is there no sensor for it?
> 
> If not, maybe you could add an input field for the total capacity, and the add a sensor to caluclate the kWh based on current SOC?

Hi,

I added the following YAML code to my configuration.yaml for my 5KW battery and substracted 20% (= 1000 Watt) as this amount is not useable. It will create a new entity.

`# Solar Batterie: Restkapazität in Watt
  - platform: template
    sensors:
      solarman_battery_soc:
        friendly_name: "Solar Batterie Restkapazität"
        unique_id: Solar_Batterie_Restkapazität
        unit_of_measurement: "W"
        value_template: "{{ (states('sensor.solarman_batter `

[image]

Hope this helps,
Joe

membersound, 2023-05-18:
What means `tel:`? Whan I use this, I get:
`Invalid config for [sensor.template]: invalid template (TemplateSyntaxError: expected token ',', got ':') for dictionary value @ data['sensors']`

PV-Joe, 2023-05-18:
> What means `tel:`? Whan I use this, I get:
> `Invalid config for [sensor.template]: invalid template (TemplateSyntaxError: expected token ',', got ':') for dictionary value @ data['sensors']`

… I’m sorry for the faulty code. Don’t know how that slipped in.

I corrected the code in the original post.
Please try again and make sure you use your entity.