StephanJoubert/home_assistant_solarman #299: Connection dropping to Deye 8KW Single-phase inverter. since version 1.4.3

cartman-sp69, 2023-05-18:
Good day, since version 1.4.1 does not work on the latest HA I have updated to 1.4.3.

However, Since doing that I'm losing connection to the inverter constantly and it is causing havoc with my automation.

This did not happen once on version 1.4.1.

Wifi signal is strong, and the Wifi Device firmware LSW3_15_FFFF_1.0.92R

Any assistance would be greatly appreciated.

DHLF, 2023-05-18:
You may check if it is the integration or the inverter who causes the issue:

Where user and password have to be changed to your actual data. As well as the ip-address of the inverter in your local network:
```
curl -u user:password 192.168.1.1/status.html
```
(this will show a LOT of "garbage" but anyway it may be a great help to locate possible issues)

My setup is loosing the connection once every hour, too .. dunno why.
By using the command I am able to grep the data with no issue but HA will wait some time until it reconnects:
```
curl -u user:password 192.168.1.1/status.html  | grep "$dat ="
```
example output(redacted):
```
var webdata_alarm = "F13"; (panel missing)
var webdata_utime = "0";
var cover_mid = "-cut-";
var cover_ver = "MW3_16U_5406_1.53";
var cover_wmode = "APSTA";
var cover_ap_ssid = "zaporizhzhia";
var cover_ap_ip = "10.10.100.254";
var cover_ap_mac = "-cut-";
var cover_sta_ssid = "-cut-";
var cover_sta_rssi = "70%";
var cover_sta_ip = "192.168.x.x";
var cover_sta_mac = "-cut-";
var status_a = "0";
var status_b = "0";
var status_c = "0";
```

I went crazy about NOT connecting to the inverter until i tried this command.. to figure it out: mistyped passwords ..

cartman-sp69, 2023-05-18:
SO nowhere do I have to option to add a username and a password, I have now tried increasing the time delta to 60 sec instead of 30 to see if it makes a difference perhaps it is trying to update too often.  I have also not changed any passwords since updating so it should not be that, when I run the above command it connects successfully and disconnects about 2 to 3 times a day.

cartman-sp69, 2023-05-18:
Also in Hassio I get the following error Querying inverter failed on connection start with exception [TimeoutError]
Update of sensor.deye_pv1_power is taking over 10 seconds