StephanJoubert/home_assistant_solarman #300: More inverters

carstenit, 2023-05-18:
I cannot get it to work with more than one inverter,  i believe it could be a json error - dublicate key “name" or something like that?

DHLF, 2023-05-18:
To add another inverter you MUST add another solarman integration.
DONT change the (first) one you got already running! Any change there will "drop" the first inverter and add the second!

Instead click on the gear(spiked circle) on the left(Configuration/Settings?)
The next page should be the one with all already found integrations like "sun", your solarman ..
Click the button in the lower right "add integration" and select AGAIN solarman.
Configure this to your needs.

Strange description due to other language settings. sorry.

DHLF, 2023-05-25:
I got a master-slave configuration working(mixed 600+2000).. where the 600 as first and the 2000 as last device in the "string" so output of the 2000 would went into the net.
*may NOT be as intended by manufacturer(mixing microinverters of different types) * so dont try this!!
I did NOT change the inverter slave ids.. so they stayed both at 1.

In this configuration both showed up with different "prefixes" in the entity names.

---
Things you may try:
If you mistyped the inverter_serial in the HA config/solarman it will NEVER be able to connect therefore unknown values so doublecheck this.
MAYBE it is right to switch the slave id to 2 but as stated above it worked without changing this parameter

NikiSkaarup, 2023-06-11:
I finally got my second inverter working after some trial and error

So in order to get more than 1 inverter to work I had to make a copy of `deye_sg04lp3.yaml` as `deye_sg04lp3_no_bat.yaml` file and for the secondary inverter and add it in the `const.py` file so I could select it

in the `deye_sg04lp3_no_bat.yaml` file find
```yaml
      - name: "Battery Temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.1
        rule: 1
        offset: 1000
        registers: [0x024A]
        icon: "mdi:battery"
        validation:
          min: 1
          max: 99
          invalidate_all:
```
and remove
```yaml
        validation:
          min: 1
          max: 99
          invalidate_all:
```

since my second inverter isn't connected directly to the batteries, it reports battery temp as -100, so it failed the validation and invalidated all fields

b1lei, 2023-06-12:
> I finally got my second inverter working after some trial and error
>
> So in order to get more than 1 inverter to work I had to make a copy of `deye_sg04lp3.yaml` as `deye_sg04lp3_no_bat.yaml` file [...]

Worked for me. 
Thanks

djsamlb, 2023-06-20:
Finally!!
Worked for me too
Thank u!!

robert10055, 2023-09-22:
Great
it works perfectly

dlx295, 2023-11-09:
Hi,

I added a new Deye Inverter to my system an want to add it to home assistant via Solarman integration.... unfortunately it is not working... the second inverter is just showing unknown values. did everything (also with changes to const.py) but it is not working at all... 

[image]

robert10055, 2023-11-09:
You must here setup number 1
[image]

dlx295, 2023-11-09:
thank you for your answer, but I do not see any changes when I use 1

NikiSkaarup, 2023-11-10:
@dlx295 there might be more than one value that has

```
   invalidate_all:
```

which causes all values to become unknown

dlx295, 2023-11-10:
> @dlx295 there might be more than one value that has
> 
> ```
>    invalidate_all:
> ```
> 
> which causes all values to become unknown

where do I have to search for it and how can I solve it? thanks

ornulf, 2024-04-07:
> Worked for me. Thanks

Worked like a charm for me .

Niklas1E, 2024-09-18:
> I finally got my second inverter working after some trial and error
>
> So in order to get more than 1 inverter to work I had to make a copy of `deye_sg04lp3.yaml` as `deye_sg04lp3_no_bat.yaml` file [...]

Hi, 
i also have 2x the DEYE SUN 12K SG04.
The master displays all data.
The slave is “unknown”.
I have also created a 2nd YAML and removed the values, then restarted HA and assigned the new YAML to the slave. 
[image]
Unfortunately without success for me.
[image]

What exactly did you do in the const.py?

Would anyone like to help me?
Thank you very much!

b1lei, 2024-09-18:
> Hi, i also have 2x the DEYE SUN 12K SG04. The master displays all data. The slave is “unknown”. I have also created a 2nd YAML and removed the values, then restarted HA and assigned the new YAML to [...]

Hi,

Try mine configs. 
[deye_sg04lp3_no_bat.yaml.zip](https://github.com/user-attachments/files/17044516/deye_sg04lp3_no_bat.yaml.zip)
[deye_sg04lp3.yaml.zip](https://github.com/user-attachments/files/17044518/deye_sg04lp3.yaml.zip)