StephanJoubert/home_assistant_solarman #301: Write Modbus Registers?

membersound, 2023-05-18:
Is it possible or will it be possible to write to the inverters modbus registers?

I want to be able to write to: 212 - BATT CHARGE LIMIT CURRENT
Is this in theory possible over the used TCP connection in this integration?
And if so, might there be a writing function in future?

qfinucane, 2023-05-21:
Check this thread, modbus writes being done through node red.

https://powerforum.co.za/topic/8451-sunsynk-inverter-monitoring/

KKES78, 2023-05-29:
> Check this thread, modbus writes being done through node red.
> 
> https://powerforum.co.za/topic/8451-sunsynk-inverter-monitoring/

Doesn't the node-red flow require a cabled connection?

KKES78, 2023-05-29:
> Is it possible or will it be possible to write to the inverters modbus registers?
> 
> I want to be able to write to: 212 - BATT CHARGE LIMIT CURRENT Is this in theory possible over the used TCP connection in this integration? And if so, might there be a writing function in future?

Agree, it would be great to be able to write to selected registries e.g. Solar Sell, now that we are seeing more and more frequently that you actually have to pay for exporting power at times. This is becoming a problem in Denmark, Germany and the Netherlands. Probably also in other countries.

membersound, 2023-05-29:
I switched to the following integration that supports writing through a esp32 module:
https://github.com/klatremis/esphome-for-deye

Anyways, it would be great it writing could be supported simply through the TCP modbus registers, without having to install the esp module.

qfinucane, 2023-05-30:
Found this, also uses the solarmanV5 integration
https://github.com/githubDante/deye-controller

theyo-tester, 2023-09-21:
For this to be possible, the inverter has to support it in the first place. For instance, as I see (and interpret) in a Modbus reference document for Deye deye_sg04lp3, only Modbus Functioncode 10h ist supported right now, meaning, writing multiple registers at once. 
Now the question goes to the authors/devs (@StephanJoubert, @jmccrohan ?)  of this repo: does the component understand and can it handle an instruction mb_functioncode: 0x10 that would be defined in the yaml file?

Thanks for the great code!

theyo-tester, 2023-09-21:
I have to add an important EDIT:

The last version of this integration has an updated service option: **Write Multiple Holding Registers (Modbus Function Code 16)**

I could change the f.i. the max A Charge current (register 108 for my model) without any issue. 

Now I can create automations where I can change the charge current depending on season and the battery SOC!

Great! 👍

KKES78, 2023-09-23:
> **NEVERMIND,** important news:
> 
> The last version of this integration has an updated service option: **Write Multiple Holding Registers (Modbus Function Code 16)** (whereas, 16dec is exactly the early mentioned mb_functioncode: 0x10hex)
> 
> I could change the f.i. the max A Charge current (register 108 for my model, you have to specify it as decimal value) without any issue.
> 
> Now I can create automations where I can change the charge current depending on season and the battery SOC! 🎉
> 
> Great! 👍

Thanks a lot for the tip. I updated and I do see the service and have set up a simple automation to disable Solar Sell when prices are negative. However, the automation is unable to update the Solar Sell toggle on my Hybrid Deye 12 kW inverter. I am also unable to update the Charge current. I have not tried other registers. I am using mod register 145 for Solar Sell and am setting the value to 0. Should this not disable the feature? For Charge current I am trying to update mod register 108 also to no avail.

Any ideas? Are there prerequisites for using the Service which must installed first?

theyo-tester, 2023-09-25:
Do you have the same inverter as mine (DEYE/Sunsynk/SolArk Hybrid 8/12K-SG04LP3) ? Different inverters have different registers. If it is one of these then....:
What values are you using for instance for register 108? The value has to be in Amper and I tested only with absolute values.
Maybe also an outdated firmware may be the reason 🤔? Are you able to remote control your inverter over the app?

KKES78, 2023-09-25:
> Do you have the same inverter as mine (DEYE/Sunsynk/SolArk Hybrid 8/12K-SG04LP3) ? Different inverters have different registers. If it is one of these then....: What values are you using for [...]

I have the 12 kW SG04LP3-EU.
Updating values remotely via Solarman Business works fine.
For register 108 I tried updating to "80". I currently have 90 A as the max charge current.
This is my versioning information:
Protocol Version：0104
MAIN：2005-1124-1807
HMI：1001-C031

KKES78, 2023-09-30:
Updating the values is working fine after reinstalling HA. This is fantastic! Big thanks to all working on this.

theyo-tester, 2023-09-30:
Great to hear that!
Sorry, I was busy last week and I also did not have any other hint for your issue.....

NikolajChristensen, 2023-10-01:
> Updating the values is working fine after reinstalling HA. This is fantastic! Big thanks to all working on this.
Nice work. Could you share some more details about how to do that? Thanks in advance.

otello86, 2023-10-03:
Hi,
i'm interested to use this feature to force charge mode of the battery.
How / where can i find the different register for my inverter Sofar HYD-5000ES ?

Elijen, 2024-04-02:
> Hi, i'm interested to use this feature to force charge mode of the battery. How / where can i find the different register for my inverter Sofar HYD-5000ES ?

Same question here. Would like to change some values but no idea where to find register addresses.

otello86, 2024-04-03:
> > Hi, i'm interested to use this feature to force charge mode of the battery. How / where can i find the different register for my inverter Sofar HYD-5000ES ?
> 
> Same question here. Would like to change some values but no idea where to find register addresses.

i don't find these register and now i use sofar2mqtt which perfectly worked

qfinucane, 2024-04-03:
Sofar Modbus register list here - https://community.openhab.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf

It does not explicitly give a list of inverters that its applicable to, therefore I would advise first querying the register and see if you get the correct data response.

voltagoo, 2025-04-10:
@qfinucane how can I query/read the registers? Only Write Holding Register is available under dev tools -> services?

qfinucane, 2025-04-10:
> [@qfinucane](https://github.com/qfinucane) how can I query/read the registers? Only Write Holding Register is available under dev tools -> services?

@voltagoo , Apologies, I don't think understand your question. The Solarman app is used to read the registers. Within the config (Inverter_definitions) you select a config file based on the inverter you are using. You can customise the registers in the file if you want. I'm not sure if that answers your question. ( sorry I don't know what "dev tools -> services" is, is this within Home assistant?)