StephanJoubert/home_assistant_solarman #309: PV production Sensor showing values at nighttime

PV-Joe, 2023-05-26:
Lately my Solarman production sensor shows values at nighttime.

Inverter: Deye 8K Hybrid
Inverter Firmware: 2005-1124-1807
HA Solarman version: v1.4.3
Logger Module Version: LSW_15_FFFF_1.092R
Logger Extended Version: v1.1.00.0F

[image]

Is there any chance to avoid this behavior?
Any suggestions are highly appreciated!

cartman-sp69, 2023-05-27:
This is because it is disconnecting with the integration or something this started happening on version v1.4.3 I'm sitting with the same issue I downgraded to version 1.4.1 it resolved the issue but 1.4.1 does not work on the latest version of home assistant I also have an issue open regarding this.

PV-Joe, 2023-05-27:
It never happened with Solarman 1.4.1. as well.

Today my “Total Load Power” went nuts showing me some 65.000 Watts. 😳

Please fix this issue.

zkrige, 2023-06-06:
I dont know much about hassio plugins, but there should be a way to smooth data and ignore temporary spikes?

david-bann, 2023-06-27:
I have the same problem with Deye 12K Hybrid 3-phase inverter...

joanguixa2, 2023-06-30:
Mismo problema con microinversor 
deye_4mppt 2000. No puedo utilizar para medir la produccion solar diaria, ya que me suma los datos del día anterior.