StephanJoubert/home_assistant_solarman #311: K Star + SolarmanPV

thhomesystem, 2023-05-28:
Hi, I have solar panels and a battery from Kstar. I use SolarmanPV, I see all the info via the app and website. I am trying to make the home_assistant_solarman work via HA. I have linked the IP as described in the manual, but I do not receive any data. Is there anyone who can help me on my way?

Thanks in advance , Tom

frsantos, 2023-05-28:
Hello @thhomesystem 

There is a new inverter definition for KStar in https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/kstar_hybrid.yml, but it has not been releaset yet. The PR is [here](https://github.com/StephanJoubert/home_assistant_solarman/pull/253) in case you want to read all discussion about it.

You may try it selecting "custom_parameters.yaml" and manually pushing the inverter file.

thhomesystem, 2023-09-12:
Yes , it is working now, last update did thé trick