StephanJoubert/home_assistant_solarman #320: How do I get my system to give information?

Pargey79, 2023-06-05:
I get unknown on every item?[image]

Joshuawo1, 2023-06-05:
Evening had a similar problem. message me on zero 84 five seven 4 six 9 seven nine. let me try assist you

ChristophCaina, 2023-06-09:
1. what Inverter are you using?
2. what yaml file have you selected during the configuration?

to support you, you need to provide a few more information :)

Pargey79, 2023-06-09:
I have a sofar solar inverter 3600 and I have selected the one for that inverter on the yaml?

Thanks for the reply.

Kind Regards

RP

ChristophCaina, 2023-06-09:
Usually, If you get 'unknown' values, this indicates that you are using the wrong yaml Template.

[image]

Since I don't have such an Inverter, I cannot tell you, which File the correct one would be, but I would try the different ones that are available until you get the values.

lordvalium, 2023-08-09:
I have the 12kW 3phase Deye. The Yaml file "deye_hybrid" does obviosuly not work as i get no correct information back. 

Can it be because my serial number starts with 27...?

PV-Joe, 2023-08-10:
Choose deye_sg04lp3.yaml and restart Home Assistant. This works with my Deye 8K Hybrid.