StephanJoubert/home_assistant_solarman #323: I use a Deye hybrid inverter whose serial number starts with 27xxx

hd00842, 2023-06-10:
My serial number starts with 271xxxx . I need how to add device to Hass please help me. What custom components do I need to use?

picardi21, 2023-06-14:
Anhe io ho un inverter con numero di serie che inzia con 271xxxx, hai risolto? a me legge l'entità ma i valori sono sconosciuti.

PV-Joe, 2023-06-16:
> Anhe io ho un inverter con numero di serie che inzia con 271xxxx, hai risolto? a me legge l'entità ma i valori sono sconosciuti.

Translation:
I also have an inverter with a serial number starting with 271xxxx, have you solved it? to me it reads the entity but the values ​​are unknown.

Please use English language 🙏

MarmaGoms, 2023-07-01:
Please double check that HA and logger are in the same network. Just close case #314 as this was a problem in local network configuration. There also the logger was 27xxx series.

Edpcomms, 2023-07-18:
I have the same issue, but both HA and Logger are on the same network.
HA 192.168.0.234 and Logger 192.168.0.245. I cannot ping logger 192.168.0.245 and cannot open logger http://192.168.0.245

lordvalium, 2023-08-09:
> Translation: I also have an inverter with a serial number starting with 271xxxx, have you solved it? to me it reads the entity but the values ​​are unknown.
>
> Please use English language 🙏

Same here, but my devices are in the same network!